simplefoc/Arduino-FOC #177: A small issue about the latest library(2.2.2)

techHappyWater, 2022-05-06:
I update the library to the latest one, and in openloop velocity control, when I set the target speed to a positive value, such as 
"motor0.move((float)(20.0));"
the motor can work well; but when I set the target speed to a negative value, such as 
"motor0.move((float)(-20.0));"
the motor can't retate. 

I don't know if my code is wrong. but In V2.1.1, it can work well and can rotate in the opposite direction.

askuric, 2022-05-06:
Huh that is an interesting one!
If you are refering to the v2.2.1 vs v2.2.2 for the open loop there is only one chnage: added possibility to limit the current instead of the voltage using the motors KV rating and the phase resistance.

In terms of direction of the target velocity the code did not change :D

I am rally not sure what might be the issue in your case, I'll double check on in my setups as well just to be sure that we did not overlook something.

Can you give  us a bit more info about your setup?
And maybe show us the code?

techHappyWater, 2022-05-06:
> Huh that is an interesting one! If you are refering to the v2.2.1 vs v2.2.2 for the open loop there is only one chnage: added possibility to limit the current instead of the voltage using the motors [...]

This is my code:
#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <SimpleFOC.h>

// BLDCMotor motor0 = BLDCMotor(11, 0.15, 360);
BLDCMotor motor0 = BLDCMotor(11, 0.135, 380);
BLDCDriver3PWM driver0 = BLDCDriver3PWM(32,25,33,13);

// BLDCMotor motor1 = BLDCMotor(11, 0.15, 360);
BLDCMotor motor1 = BLDCMotor(11, 0.135, 380);
BLDCDriver3PWM driver1  = BLDCDriver3PWM(26,14,27,12);

void setup() {

  /*************************motor0 start*************************/
  motor0.controller = MotionControlType::velocity_openloop;
  motor0.foc_modulation = FOCModulationType::SinePWM;   // 默认就为正弦波
  driver0.voltage_power_supply = 12.6;
  driver0.init();
  motor0.linkDriver(&driver0);
  motor0.current_limit = 3.45; // Amps
  // motor0.voltage_limit = 1.0;   // [V]
  motor0.velocity_limit = 30; // [rad/s]
  motor0.init();
  /*************************motor0 end*************************/

  /*************************motor1 start*************************/
  motor1.controller = MotionControlType::velocity_openloop;
  motor1.foc_modulation = FOCModulationType::SinePWM;   // 默认就为正弦波
  driver1.voltage_power_supply = 12.6;
  driver1.init();
  motor1.linkDriver(&driver1);
  motor1.current_limit = 3.05; // Amps
  // motor1.voltage_limit = 1.0;   // [V]
  motor1.velocity_limit = 30; // [rad/s]
  motor1.init();
  /*************************motor1 end*************************/

  Serial.begin(115200);
  Serial.println("Motor ready!");
  _delay(1000);
}

void loop() {
  motor0.move((float)(20.0));
  motor1.move((float)(20.0));
}

when I set the motor0's target velocity to "-20", the motor will not move
And the code is modified from my previous project, use V2.1.1, instead of V2.2.1
If you can help me find whta's wrong in my project, I would be very thankful.

askuric, 2022-06-18:
Hey @techHappyWater,
What is the mcu you are using?

techHappyWater, 2022-06-18:
> Hey @techHappyWater, What is the mcu you are using?

I use esp32, The specific model is ESP32-PICO-D4

askuric, 2022-06-18:
Have you tried removing the phase resistance and KV rating parameters? 
And then setting the voltage limits?

Your code would look like this:
```cpp
#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <SimpleFOC.h>

BLDCMotor motor0 = BLDCMotor(11); //, 0.15, 360);
BLDCDriver3PWM driver0 = BLDCDriver3PWM(32,25,33,13);

BLDCMotor motor1 = BLDCMotor(11); //, 0.15, 360);
BLDCDriver3PWM driver1  = BLDCDriver3PWM(26,14,27,12);

void setup() {

  /*************************motor0 start*************************/
  motor0.controller = MotionControlType::velocity_openloop;
  motor0.foc_modulation = FOCModulationType::SinePWM;   // 默认就为正弦波
  driver0.voltage_power_supply = 12.6;
  driver0.init();
  motor0.linkDriver(&driver0);
  //motor0.current_limit = 3.45; // Amps
  motor0.voltage_limit = 1.0;   // [V]
  motor0.velocity_limit = 30; // [rad/s]
  motor0.init();
  /*************************motor0 end*************************/

  /*************************motor1 start*************************/
  motor1.controller = MotionControlType::velocity_openloop;
  motor1.foc_modulation = FOCModulationType::SinePWM;   // 默认就为正弦波
  driver1.voltage_power_supply = 12.6;
  driver1.init();
  motor1.linkDriver(&driver1);
  // motor1.current_limit = 3.05; // Amps
  motor1.voltage_limit = 1.0;   // [V]
  motor1.velocity_limit = 30; // [rad/s]
  motor1.init();
  /*************************motor1 end*************************/

  Serial.begin(115200);
  Serial.println("Motor ready!");
  _delay(1000);
}

void loop() {
  motor0.move((float)(20.0));
  motor1.move((float)(20.0));
}
```

If the phase resistance and kv are not set well it can happen that your motor does not move well. 
You can also try raising the phase resistance value from 0.15 to 0.5. This will raise the current sent to the motor and give it a bit more torque. 
Could you verify if the motor spins in both directions with one of these changes?

techHappyWater, 2022-06-18:
> Could you verify if the motor spins in both directions with one of these changes?

OK, thank you for your opinion. I don't have the driver board now. I'll test it when I get it

techHappyWater, 2022-08-02:
> Have you tried removing the phase resistance and KV rating parameters? And then setting the voltage limits?
> 
> Your code would look like this:
> 
> ```c++
> #include <Arduino.h>
> #include <SPI.h>
> #include <Wire.h>
> #include <SimpleFOC.h>
> 
> BLDCMotor motor0 = BLDCMotor(11); //, 0.15, 360);
> BLDCDriver3PWM driver0 = BLDCDriver3PWM(32,25,33,13);
> 
> BLDCMotor motor1 = BLDCMotor(11); //, 0.15, 360);
> BLDCDriver3PWM driver1  = BLDCDriver3PWM(26,14,27,12);
> 
> void setup() {
> 
>   /*************************motor0 start*************************/
>   motor0.controller = MotionControlType::velocity_openloop;
>   motor0.foc_modulation = FOCModulationType::SinePWM;   // 默认就为正弦波
>   driver0.voltage_power_supply = 12.6;
>   driver0.init();
>   motor0.linkDriver(&driver0);
>   //motor0.current_limit = 3.45; // Amps
>   motor0.voltage_limit = 1.0;   // [V]
>   motor0.velocity_limit = 30; // [rad/s]
>   motor0.init();
>   /*************************motor0 end*************************/
> 
> 
>   /*************************motor1 start*************************/
>   motor1.controller = MotionControlType::velocity_openloop;
>   motor1.foc_modulation = FOCModulationType::SinePWM;   // 默认就为正弦波
>   driver1.voltage_power_supply = 12.6;
>   driver1.init();
>   motor1.linkDriver(&driver1);
>   // motor1.current_limit = 3.05; // Amps
>   motor1.voltage_limit = 1.0;   // [V]
>   motor1.velocity_limit = 30; // [rad/s]
>   motor1.init();
>   /*************************motor1 end*************************/
> 
> 
>   Serial.begin(115200);
>   Serial.println("Motor ready!");
>   _delay(1000);
> }
> 
> void loop() {
>   motor0.move((float)(20.0));
>   motor1.move((float)(20.0));
> }
> ```
> 
> If the phase resistance and kv are not set well it can happen that your motor does not move well. You can also try raising the phase resistance value from 0.15 to 0.5. This will raise the current sent to the motor and give it a bit more torque. Could you verify if the motor spins in both directions with one of these changes?

I tried your code, it works correct! Thank you very much!

askuric, 2022-10-15:
related to #219

closed 2022-10-15 by askuric