simplefoc/Arduino-FOC #184: [FEATURE]Cache the results of trigonometric functions？

FFiot, 2022-05-23:
st = _sin(angle);
ct = _cos(angle);
Cache the results of trigonometric functions? Angle does not change between samples。
Clarke transform：src\BLDCMotor.cpp Line 546.  
Inverse clarke transform：src\common\base_classes\CurrentSense.cpp Line 42.

// function approximating cosine calculation by using fixed size array
// ~55us (float array)
// ~56us (int array)

runger1101001, 2022-05-24:
Makes sense. This will also help people who replace the trig functions with more accurate but potentially slower versions.

arturohernandez10, 2022-05-24:
If a change is done here, it may be a good idea to better provide a way to use a table of values. Depending on the resolution, the actual could be extrapolated or looked up. That would allow people to use different kinds of waves. Perhaps, parametrizing by current or distance to the pole...

runger1101001, 2022-05-25:
Um, that's actually the way the trig functions already work...

_sin() and _cos() are defined in https://github.com/simplefoc/Arduino-FOC/blob/master/src/common/foc_utils.cpp
They use a lookup table.

But I agree it could be cool to allow the user more control over this lookup table, or to replace the _sin() and _cos() with their own, perhaps MCU-optimised implementations.

runger1101001, 2023-03-21:
https://github.com/simplefoc/Arduino-FOC/pull/262

runger1101001, 2023-03-21:
https://community.simplefoc.com/t/embedded-world-2023-stm32-cordic-co-processor/3107