simplefoc/Arduino-FOC #188: Lowside current sense initialization causes ESP32 Core Panic exceptions in 2.2.2 version

kimiolek, 2022-05-30:
Hi,
I'm playing with simplefoc lib and lowside current sense a little bit.
I add a current sense oject to my code like 
`LowsideCurrentSense cs = LowsideCurrentSense(0.015f, 20.0f, CS_A, CS_B);`
and in setup() call init method 
`cs.init();`
the esp32 get core panic exception, exactly in this line, when I comment it no problem exists.

Serial monitor output:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3d90  PS      : 0x00060830  A0      : 0x800d1500  A1      : 0x3ffb27d0  
A2      : 0x3ffc1440  A3      : 0x3ffc1480  A4      : 0x0800001c  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb27a0  
A10     : 0x3ffb9115  A11     : 0x3ff5e000  A12     : 0x00000000  A13     : 0xffffcfc7  
A14     : 0x000061a8  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000020  LBEG    : 0x40085f58  LEND    : 0x40085f63  LCOUNT  : 0x00000000  

Backtrace:0x400d3d8d:0x3ffb27d00x400d14fd:0x3ffb27f0 0x400d5622:0x3ffb2820 
ELF file SHA256: 0000000000000000
Rebooting...
```
The problem ocurres in version 2.2.2 of simplefoc. In 2.2.1 it seems to be good, there is no core panic error.

**Hardware setup:**
- ESP-32 ESP-WROOM-32 board
- own motor controller board, but exception ocurres when I use only ESP32 board too - anything disconected.

**IDE**
- Arduino IDE and PlatformIO - both the same exception
- esp32 package in 2.0.3 version by espressif

askuric, 2022-05-31:
Hey @kimiolek,

Maybe a stupid question but did you link the driver and the current sense:
```
cs.linkDriver(&driver);
```

micology-code, 2022-06-30:
Same thing happen to me
PlatformIO Core 6.0.2·Home 3.4.2

platformio.ini
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = askuric/Simple FOC@^2.2.2
lib_archive = false
monitor_filters = esp32_exception_decoder
monitor_speed = 115200
build_type = debug
```

Code
```
#include <SimpleFOC.h>

InlineCurrentSense current_sense = InlineCurrentSense(0.01, 50.0, 33, 35);

void setup() {
  Serial.begin(115200);

  current_sense.init();
  
  Serial.println("Current sense ready.");
}

void loop() {
}
```

Serial output
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d1f5d  PS      : 0x00060830  A0      : 0x800d221c  A1      : 0x3ffb27d0
A2      : 0x3ffc1448  A3      : 0x3ffbdbec  A4      : 0x0800001c  A5      : 0x00000003
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb2770
A10     : 0x00000001  A11     : 0x00000021  A12     : 0x00000023  A13     : 0xffffcfc7
A14     : 0x3ffb7e48  A15     : 0x00000000  SAR     : 0x00000003  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000020  LBEG    : 0x40085e78  LEND    : 0x40085e83  LCOUNT  : 0xffffffff

Backtrace:0x400d1f5a:0x3ffb27d00x400d2219:0x3ffb27f0 0x400d3ae7:0x3ffb2820
  #0  0x400d1f5a:0x3ffb27d00 in InlineCurrentSense::init() at .pio/libdeps/esp32dev/Simple FOC/src/current_sense/InlineCurrentSense.cpp:25
```

askuric, 2022-07-08:
@micology-code  
Thanks for this you are absolutely right. This was a bug of our implementation. 
We've updated the library and the new release will have the fix. For now it is in the dev.

closed 2022-10-15 by askuric