simplefoc/Arduino-FOC #198: [BUG] ESP32 S3 compiles error in Platformio

huigang39, 2022-06-27:
**Describe the bug**
ESP32 S3 compiles error in Platformio.
`.pio/libdeps/esp32-s3-devkitc-1/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_adc_driver.cpp:31:23: error: 'SENS_SAR_START_FORCE_REG' was not declared in this scope`

**IDE you are using**

- Platformio
- 
[image]

askuric, 2022-06-28:
Hey @huigang39,
Which version of th aruduino-esp32 package are you using?
Is is 2.0.3?
Did you maybe try with 2.0.2?

huigang39, 2022-06-28:
> Hey @huigang39, Which version of th aruduino-esp32 package are you using? Is is 2.0.3? Did you maybe try with 2.0.2?

Yes, 2.0.3 The latest stable version.
2.0.2 doesn't seem to support ESP32 S3 DevkitC-1.

huigang39, 2022-06-28:
> Hey @huigang39, Which version of th aruduino-esp32 package are you using? Is is 2.0.3? Did you maybe try with 2.0.2?

I have solved this problem, I replaced the original code in soc/sens_reg.h with the code in arduino-esp32, the code is as follows:

[image]

```c
// Copyright 2015-2016 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at

//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.
#ifndef _SOC_SENS_REG_H_
#define _SOC_SENS_REG_H_

#include "soc.h"
#define SENS_SAR_READ_CTRL_REG (DR_REG_SENS_BASE + 0x0000)
/* SENS_SAR1_DATA_INV : R/W ;bitpos:[28] ;default: 1'd0 ; */
/*description: Invert SAR ADC1 data*/
#define SENS_SAR1_DATA_INV (BIT(28))
#define SENS_SAR1_DATA_INV_M (BIT(28))
#define SENS_SAR1_DATA_INV_V 0x1
#define SENS_SAR1_DATA_INV_S 28
/* SENS_SAR1_DIG_FORCE : R/W ;bitpos:[27] ;default: 1'd0 ; */
/*description: 1: SAR ADC1 controlled by DIG ADC1 CTRL  0: SAR ADC1 controlled by RTC ADC1 CTRL*/
#define SENS_SAR1_DIG_FORCE (BIT(27))
#define SENS_SAR1_DIG_FORCE_M (BIT(27))
#define SENS_SAR1_DIG_FORCE_V 0x1
#define SENS_SAR1_DIG_FORCE_S 27
/* SENS_SAR1_SAMPLE_NUM : R/W ;bitpos:[26:19] ;default: 8'd0 ; */
/*description: */
#define SENS_SAR1_SAMPLE_NUM 0x000000FF
#define SENS_SAR1_SAMPLE_NUM_M ((SENS_SAR1_SAMPLE_NUM_V) << (SENS_SAR1_SAMPLE_NUM_S))
#define SENS_SAR1_SAMPLE_NUM_V 0xFF
#define SENS_SAR1_SAMPLE_NUM_S 19
/* SENS_SAR1_CLK_GATED : R/W ;bitpos:[18] ;default: 1'b1 ; */
/*description: */
#define SENS_SAR1_CLK_GATED (BIT(18))
#define SENS_SAR1_CLK_GATED_M (BIT(18))
#define SENS_SAR1_CLK_GATED_V 0x1
#define SENS_SAR1_CLK_GATED_S 18
/* SENS_SAR1_SAMPLE_BIT : R/W ;bitpos:[17:16] ;default: 2'd3 ; */
/*description: 00: for 9-bit width  01: for 10-bit width  10: for 11-bit width
  11: for 12-bit width*/
#define SENS_SAR1_SAMPLE_BIT 0x00000003
#define SENS_SAR1_SAMPLE_BIT_M ((SENS_SAR1_SAMPLE_BIT_V) << (SENS_SAR1_SAMPLE_BIT_S))
#define SENS_SAR1_SAMPLE_BIT_V 0x3
#define SENS_SAR1_SAMPLE_BIT_S 16
/* SENS_SAR1_SAMPLE_CYCLE : R/W ;bitpos:[15:8] ;default: 8'd9 ; */
/*description: sample cycles for SAR ADC1*/
#define SENS_SAR1_SAMPLE_CYCLE 0x000000FF
#define SENS_SAR1_SAMPLE_CYCLE_M ((SENS_SAR1_SAMPLE_CYCLE_V) << (SENS_SAR1_SAMPLE_CYCLE_S))
#define SENS_SAR1_SAMPLE_CYCLE_V 0xFF
#define SENS_SAR1_SAMPLE_CYCLE_S 8
/* SENS_SAR1_CLK_DIV : R/W ;bitpos:[7:0] ;default: 8'd2 ; */
/*description: clock divider*/
#define SENS_SAR1_CLK_DIV 0x000000FF
#define SENS_SAR1_CLK_DIV_M ((SENS_SAR1_CLK_DIV_V) << (SENS_SAR1_CLK_DIV_S))
#define SENS_SAR1_CLK_DIV_V 0xFF
#define SENS_SAR1_CLK_DIV_S 0

#define SENS_SAR_READ_STATUS1_REG (DR_REG_SENS_BASE + 0x0004)
[... 975 lines cut ...]
/* SENS_SAR1_DAC_XPD_FSM : R/W ;bitpos:[3:0] ;default: 4'b0011 ; */
/*description: */
#define SENS_SAR1_DAC_XPD_FSM 0x0000000F
#define SENS_SAR1_DAC_XPD_FSM_M ((SENS_SAR1_DAC_XPD_FSM_V) << (SENS_SAR1_DAC_XPD_FSM_S))
#define SENS_SAR1_DAC_XPD_FSM_V 0xF
#define SENS_SAR1_DAC_XPD_FSM_S 0

#define SENS_SAR_NOUSE_REG (DR_REG_SENS_BASE + 0x00F8)
/* SENS_SAR_NOUSE : R/W ;bitpos:[31:0] ;default: 32'h0 ; */
/*description: */
#define SENS_SAR_NOUSE 0xFFFFFFFF
#define SENS_SAR_NOUSE_M ((SENS_SAR_NOUSE_V) << (SENS_SAR_NOUSE_S))
#define SENS_SAR_NOUSE_V 0xFFFFFFFF
#define SENS_SAR_NOUSE_S 0

#define SENS_SARDATE_REG (DR_REG_SENS_BASE + 0x00FC)
/* SENS_SAR_DATE : R/W ;bitpos:[27:0] ;default: 28'h1605180 ; */
/*description: */
#define SENS_SAR_DATE 0x0FFFFFFF
#define SENS_SAR_DATE_M ((SENS_SAR_DATE_V) << (SENS_SAR_DATE_S))
#define SENS_SAR_DATE_V 0xFFFFFFF
#define SENS_SAR_DATE_S 0

#endif /*_SOC_SENS_REG_H_ */
```

askuric, 2022-06-29:
Hey @huigang39,

ESPs3 does not have the same sens_reg.h file its true. 
These sens reg registers are used in the esp32 adc driver that I've reused from an old esp32 arduino package version, I think they've used it until 1.6.0 version. The reason is the speed, one `analogRead` takes almost 50us on ESP32 2.0. and the old one (now in `adcRead`) takes only 10.

As ESP32S3 does not have the same registers in sens_reg.h I would have said that there is nothing you can really do except to delete the driver `esp32_adc_driver.c/h` and use `analogRead` instead.

When you replaced the code, were you able to use current control of you motor?
Are current values making sesnse, they are read well?

huigang39, 2022-07-04:
> 嘿@huigang39，
> 
> ESP3 没有相同的 sens_reg.h 文件。这些传感注册寄存器用于我从旧的esp32 arduino软件包版本中重用的esp32 adc驱动程序，我认为他们已经使用它直到1.6.0版本。原因是速度，在ESP32 2.0上需要近50us。而旧的（现在）只需要10。`analogRead``adcRead`
> 
> 由于 ESP32S3 在 sens_reg.h 中没有相同的寄存器，我会说，除了删除驱动程序并改用之外，您实际上无能为力。`esp32_adc_driver.c/h``analogRead`
> 
> 当您替换代码时，您是否能够使用电机的当前控制？当前的价值观是否使sesnse，它们被很好地阅读了？

I don't know how to connect the driver board with the ESP32 S3, my motor won't turn after I connect it.😥

huigang39, 2022-07-04:
> Hey @huigang39,
>
> ESPs3 does not have the same sens_reg.h file its true. These sens reg registers are used in the esp32 adc driver that I've reused from an old esp32 arduino package version, I think [...]

I don't know how to connect the driver board with the ESP32 S3, my motor won't turn after I connect it.😥

askuric, 2022-07-08:
Hey @huigang39,

I'll try to do some tests on my own and figure out a way around this. I would appreciate if you could let me know the exeuciton time of the `analogRead` of the esp32s3. 
Here is the community post: https://community.simplefoc.com/t/help-needed-esp32s3-analogread-execution-time/2221/2

Fu-Xiaotian, 2022-07-14:
> Hey @huigang39,
>
> I'll try to do some tests on my own and figure out a way around this. I would appreciate if you could let me know the exeuciton time of the `analogRead` of the esp32s3. Here is the [...]

Hi, @askuric 
I tried to modify `esp32_adc_driver.c/h`, and then the motor works normally. There are still some of the same registers on esp32s3/s2, but their names have changed. Some registers no longer exist, so I commented them out in the code.
I hope these codes can help you.

```
#include "esp32_adc_driver.h"

#if defined(ESP_H) && defined(ARDUINO_ARCH_ESP32) && defined(SOC_MCPWM_SUPPORTED) 

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "rom/ets_sys.h"
#include "esp_attr.h"
#include "esp_intr.h"
#include "soc/rtc_io_reg.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/sens_reg.h"

static uint8_t __analogAttenuation = 3;//11db
static uint8_t __analogWidth = 3;//12 bits
static uint8_t __analogCycles = 8;
static uint8_t __analogSamples = 0;//1 sample
static uint8_t __analogClockDiv = 1;

// Width of returned answer ()
static uint8_t __analogReturnedWidth = 12;

void __analogSetWidth(uint8_t bits){
    if(bits < 9){
        bits = 9;
    } else if(bits > 12){
        bits = 12;
    }
    __analogReturnedWidth = bits;
    __analogWidth = bits - 9;
    // SET_PERI_REG_BITS(SENS_SAR_START_FORCE_REG, SENS_SAR1_BIT_WIDTH, __analogWidth, SENS_SAR1_BIT_WIDTH_S);
    // SET_PERI_REG_BITS(SENS_SAR_READER1_CTRL_REG, SENS_SAR1_SAMPLE_BIT, __analogWidth, SENS_SAR1_SAMPLE_BIT_S);

    // SET_PERI_REG_BITS(SENS_SAR_START_FORCE_REG, SENS_SAR2_BIT_WIDTH, __analogWidth, SENS_SAR2_BIT_WIDTH_S);
    // SET_PERI_REG_BITS(SENS_SAR_READER2_CTRL_REG, SENS_SAR2_SAMPLE_BIT, __analogWidth, SENS_SAR2_SAMPLE_BIT_S);
}

void __analogSetCycles(uint8_t cycles){
    __analogCycles = cycles;
    // SET_PERI_REG_BITS(SENS_SAR_READER1_CTRL_REG, SENS_SAR1_SAMPLE_CYCLE, __analogCycles, SENS_SAR1_SAMPLE_CYCLE_S);
    // SET_PERI_REG_BITS(SENS_SAR_READER2_CTRL_REG, SENS_SAR2_SAMPLE_CYCLE, __analogCycles, SENS_SAR2_SAMPLE_CYCLE_S);
}

void __analogSetSamples(uint8_t samples){
    if(!samples){
        return;
    }
    __analogSamples = samples - 1;
    SET_PERI_REG_BITS(SENS_SAR_READER1_CTRL_REG, SENS_SAR1_SAMPLE_NUM, __analogSamples, SENS_SAR1_SAMPLE_NUM_S);
    SET_PERI_REG_BITS(SENS_SAR_READER2_CTRL_REG, SENS_SAR2_SAMPLE_NUM, __analogSamples, SENS_SAR2_SAMPLE_NUM_S);
}

void __analogSetClockDiv(uint8_t clockDiv){
    if(!clockDiv){
        return;
    }
    __analogClockDiv = clockDiv;
    SET_PERI_REG_BITS(SENS_SAR_READER1_CTRL_REG, SENS_SAR1_CLK_DIV, __analogClockDiv, SENS_SAR1_CLK_DIV_S);
    SET_PERI_REG_BITS(SENS_SAR_READER2_CTRL_REG, SENS_SAR2_CLK_DIV, __analogClockDiv, SENS_SAR2_CLK_DIV_S);
}

[... 171 lines cut ...]
    }

    uint16_t value = 0;

    if(channel > 7){
        while (GET_PERI_REG_MASK(SENS_SAR_MEAS2_CTRL2_REG, SENS_MEAS2_DONE_SAR) == 0); //wait for conversion
        value = GET_PERI_REG_BITS2(SENS_SAR_MEAS2_CTRL2_REG, SENS_MEAS2_DATA_SAR, SENS_MEAS2_DATA_SAR_S);
    } else {
        while (GET_PERI_REG_MASK(SENS_SAR_MEAS1_CTRL2_REG, SENS_MEAS1_DONE_SAR) == 0); //wait for conversion
        value = GET_PERI_REG_BITS2(SENS_SAR_MEAS1_CTRL2_REG, SENS_MEAS1_DATA_SAR, SENS_MEAS1_DATA_SAR_S);
    }

    // Shift result if necessary
    uint8_t from = __analogWidth + 9;
    if (from == __analogReturnedWidth) {
        return value;
    }
    if (from > __analogReturnedWidth) {
        return value >> (from - __analogReturnedWidth);
    }
    return value << (__analogReturnedWidth - from);
}

#endif
```

askuric, 2022-07-14:
Awesome, thanks!
This is going to be a great starting point. 😄

closed 2022-10-15 by askuric