simplefoc/Arduino-FOC #200: How to deal with deadband voltage of stepper?

Zahidilyas, 2022-06-27:
**Describe the bug**
Stepper motor does not respond to voltages below 3V. On all the control methods, for voltages below 3V, the stepper does not turn, hence it is not able to track the position smoothly.

**Describe the hardware setup**
For us it is very important to know what is the hardware setup you're using in order to be able to help more directly

**Nema 17 stepper motor**
L298N
Teensy 4.1
Optical Encoder 1000PPR(4000CPR)

**IDE you are using**
Arduino IDE

runger1101001, 2022-09-19:
A similar problem exists for BLDCs using very slow switching drivers (actually like the L298N) - the driver can't fully open for low PWM duty cycles, leading to no movement. 
Such setups also have a minimum voltage to enable motion.

Another reason to have a minimum voltage is in combination with current sensing - depending on the current sensing HW, there can be a minimum supported duty cycle for low side sensing.

dekutree64, 2026-05-27:
Perhaps we should add a deadtime compensation option? It should be straightforward for stepper motors, simply increasing the duty of the active half bridge by `dead_zone * voltage_power_supply;` (there are two dead zones per PWM cycle, but the one after switch-off is supposed to be a non-conducting period anyway). The dead_zone variable currently only exists in BLDCDriver6PWM where it can be actively controlled, but it could be moved to the base class where it can be set from the datasheet value of the driver you're using.

I don't think BLDC motors can be perfectly compensated due to the interaction of the three coils, but adding `dead_zone * voltage_power_supply` to the highest phase and subtracting it from the lowest phase would be an improvement over what we have. The same procedure should give perfect compensation for hybrid stepper using BLDCDriver classes.

I tested this in BLDCDriver6PWM::setPwm and it works well.
```
  if (enable_deadtime_compensation) {
    float offset = dead_zone * voltage_power_supply;
    // If all phases equal, do nothing. If two are equal, offset both of them one way and the remaining phase the other way.
    if(Ua == Ub) { if(Ua != Uc) { if(Uc < Ua) offset = -offset; Ua -= offset, Ub -= offset, Uc += offset; } }
    else if(Ua == Uc)           { if(Ub < Ua) offset = -offset; Ua -= offset, Ub += offset, Uc -= offset; }
    else if(Ub == Uc)           { if(Ua < Ub) offset = -offset; Ua += offset, Ub -= offset, Uc -= offset; }
    // If all phases are different, offset the highest one upward and lowest one downward. The high and low phases
    // will see correct voltage to eachother, but middle phase will see less voltage to the others than it should.
    if(Ua > Ub) {
      if(Ua > Uc) { Ua += offset; if(Ub < Uc) Ub -= offset; else Uc -= offset; }
      else        { Uc += offset; if(Ua < Ub) Ua -= offset; else Ub -= offset; }
    } else {
      if(Ub > Uc) { Ub += offset; if(Ua < Uc) Ua -= offset; else Uc -= offset; }
      else        { Uc += offset; if(Ua < Ub) Ua -= offset; else Ub -= offset; }
    }
  }
```
I tested the same code in BLDCDriver3PWM on SimpleFOC Stepstick (MAX22213 driver) running a BLDC motor. The datasheet says it has 100ns dead time, but I got optimum results using 230ns. Not sure what to make of that.