simplefoc/Arduino-FOC #201: something wrong with current sense align

blacknull, 2022-07-11:
my hardware is a simple foc shield and a 3208 bldc.
after test with voltage torque control, everything seem fine. so i moved to foc current mode.
unfortunately, when i switched to foc current mode, the bldc totally lost control, it shaked like crazy and no effect with PID adjustment.
so i looked into source code and added some debug msg in 
int InlineCurrentSense::driverAlign(float voltage)
and found:

_MOT: Align current sense.
current sense gain_a: 2.000000, gian_b: -2.000000, gain_c: 2.000000
A phase current a:0.506888, b:-0.000000, c:0.000000
B phase current a:-0.000000, b:-2.774509, c:0.000000
current sense gain_a: 0.000000, gian_b: -0.000000, gain_c: 2.000000
current sense pinA: 34, pinB: 36, pinC: -12345
MOT: Success: 2_

obviosly there were something wrong with phase current datas, the sum of 3 phase current should be zero. 
after recheck the connection, i replaced all of the ina240 on simple foc shield.
but saddly nothing changed. :-(

now i have no idea about this problem, does anybody know anything?
thx!

askuric, 2022-12-22:
Hey @blacknull,
Did you manage to figure out what was causing the error?
I am not sure where do your prints come from so I am not sure what might be causing it. You could potentially paste your code here so I could try to reproduce it.

askuric, 2025-07-29:
Closing due to inactivity. Since 2022  the alignment code has been rewritten so this issue is no longer relevant.

closed 2025-07-29 by askuric