simplefoc/Arduino-FOC #215: [BUG] error: 'SIMPLEFOC_CURRENT_SENSE_INIT_FAILED' was not declared in this scope

TKroos18, 2022-09-12:
This is a simplified template, feel free to change it if it does not fit your case.

**Describe the bug**
lib\Arduino-FOC\src\current_sense\hardware_specific\stm32\stm32f4\stm32f4_mcu.cpp: In function 'void* _configureADCLowSide(const void*, int, int, int)':
lib\Arduino-FOC\src\current_sense\hardware_specific\stm32\stm32f4\stm32f4_mcu.cpp:42:75: error: 'SIMPLEFOC_CURRENT_SENSE_INIT_FAILED' was not declared in this scope
   42 |   if(_adc_init(cs_params, (STM32DriverParams*)driver_params) != 0) return SIMPLEFOC_CURRENT_SENSE_INIT_FAILED;
      |                                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

**Describe the hardware setup**
[env:genericSTM32F411CE]
platform = ststm32
board = genericSTM32F411CE
framework = arduino
upload_protocol = stlink

**IDE you are using**
- CLion + Platformio

**Tried the Getting started guide? - if applicable**
Have you tried the getting started guide and at which step are you blocked in

runger1101001, 2022-09-19:
For platformIO you should always add:
`lib_archive = false`
to platformio.ini

But this is not causing your problem.

For me, 
```
[env:genericSTM32F411CE]
platform = ststm32
board = genericSTM32F411CE
framework = arduino

lib_archive = false
```

compiles without problems on the *development* branch of SimpleFOC.

So this issue is already fixed on the dev branch. You can use this branch in your project easily, either by referencing it in the platformio.ini, or by cloning it into the lib folder of your project.

runger1101001, 2022-09-19:
Waiting for release 2.2.3 ...

closed 2022-10-15 by askuric