simplefoc/Arduino-FOC #220: [BUG]BLDCDriver3PWM driver cann't initialize using STM32duino

hp1210cl, 2022-10-05:
I am working with stm32f303cbt6 mcu under STM32duino. I created BLDCDriver3PWM  driver in arduino code. The phA is PB13, phB is PB14, phC is PB15. When I used DEBUG mode, I got the following infos:

TIM1-CH1N TIM1-CH2N TIM1-CH3N score: -3
TIM1-CH1N TIM1-CH2N TIM15-CH1N score: -3
TIM1-CH1N TIM1-CH2N TIM15-CH2 score: -3
TIM1-CH1N TIM15-CH1 TIM1-CH3N score: -3
TIM1-CH1N TIM15-CH1 TIM15-CH1N score: -3
TIM1-CH1N TIM15-CH1 TIM15-CH2 score: -3
STM32-DRV: no workable combination found on these pins
STM32-DRV: ERR: not find Best Timer Combination
MOT: Init not possible, driver not initialized

It seems that the library cann't find the best timer combination. According to the STM32F303CBT6 datasheet, TIM1_CHIN TIM1_CH2N TIM1_CH3N should be found and used as pwm output.
 When I change the phA phB phC to PA4 PB0 PB1, I got the follwing debug infos. And the motor worked fine.

TIM3-CH2 TIM1-CH2N TIM1-CH3N score: -3
TIM3-CH2 TIM1-CH2N TIM3-CH4 score: -3
TIM3-CH2 TIM1-CH2N TIM8-CH3N score: -3
TIM3-CH2 TIM3-CH3 TIM1-CH3N score: -3
TIM3-CH2 TIM3-CH3 TIM3-CH4 score: 1
TIM3-CH2 TIM3-CH3 TIM8-CH3N score: -3
TIM3-CH2 TIM8-CH2N TIM1-CH3N score: -3
TIM3-CH2 TIM8-CH2N TIM3-CH4 score: -3
TIM3-CH2 TIM8-CH2N TIM8-CH3N score: -3
STM32-DRV: best: TIM3-CH2 TIM3-CH3 TIM3-CH4 score: 1
STM32-DRV: Configuring high timer 3
STM32-DRV: Configuring high channel 2
STM32-DRV: Configuring high timer 3
STM32-DRV: Configuring high channel 3
STM32-DRV: Configuring high timer 3
STM32-DRV: Configuring high channel 4
STM32-DRV: Restarting timer 3
MOT: Init
MOT: Enable driver.

I think some bugs is hiding in the stm32_mcu.cpp, but I don't know where are they.

runger1101001, 2022-10-06:
Hi @hp1210cl ,

We don't support using the negated channels for anything except the low-side output in 6-PWM mode.

So for high-side PWM outputs in 6-PWM, and all the other modes except 6-PWM, you have to use non-negated channels (the ones without the 'N' at the end).

While it would theoretically be possible, it would add even more complexity to the already very complex stm32 PWM driver.
If someone wants to give it a try, we would consider pull-requests, but I currently have no plans to implement it myself.

Sorry! It means you have to use different pins.

fpistm, 2024-12-05:
Maybe this could help:
https://github.com/stm32duino/Arduino_Core_STM32/pull/2294

runger1101001, 2024-12-05:
@fpistm thank you very much for pointing this out and considering us! I'll take a look at it.