simplefoc/Arduino-FOC #223: [BUG] Version 2.2.3 doesn't compile for Teensy (2.2.2 does)

GeertBoer, 2022-10-21:
Simply including SimpleFoc.h 2.2.3 results in a compiler error, whilst 2.2.2 compiles just fine. Error seems to be in using type PinState?

In file included from /Users/geert/Documents/Arduino/libraries/Simple_FOC/src/SimpleFOC.h:116:0,
                 from /Users/geert/Documents/Arduino/libraries/Simple_FOC/examples/utils/communication_test/commander/commander_extend_example/commander_extend_example.ino:5:
/Users/geert/Documents/Arduino/libraries/Simple_FOC/src/communication/StepDirListener.h:56:5: error: 'PinStatus' does not name a type
     PinStatus polarity = RISING; //!< polarity of the step pin
     ^
commander_extend_example: In function 'void setup()':
commander_extend_example:37: warning: ISO C++ forbids converting a string constant to 'char*' 
     command.add('L', doLed, "led on/off");
                                         ^
commander_extend_example:38: warning: ISO C++ forbids converting a string constant to 'char*' 
     command.add('A', doAnalog, "analog read A0-A4");
                                                   ^
Fout bij het compileren voor board Teensy 3.2 / 3.1

runger1101001, 2022-10-22:
Thanks a lot for reporting this.

Unfortunately the PinStatus type is used by about half of the Arduino platforms, and not by the other half. Its a major pain.

If you want a quick fix for now, just add a
`#define PinStatus int`
to the StepDirListener.h

I'll fix it for the next release. 
We should also consider adding a Teensy build to the library tests, that would have caught this error.

runger1101001, 2022-10-30:
Fixed in dev branch.

GeertBoer, 2022-11-01:
Glad I could help

closed 2023-03-18 by askuric