simplefoc/Arduino-FOC #225: [BUG] LowsideCurrentSense.init lead to panic on ESP32-S3 DevKitC-1 board

kkksparky, 2022-10-27:
After calling in the setup of 
LowsideCurrentSense.init() 
on ESP32-S3 DevKitC-1 board it goes to panic 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Doesn't matter of the Core used. 

I have tried also with board ESP32-WROOM-32D and the behaviour is the same. 

I have used both Arduino IDE and PlatformIO with latest stable release 2.2.3

Source code:
#include <SimpleFOC.h>

#define INH_A 36
#define INH_B 38
#define INH_C 40
#define INL_A 37
#define INL_B 39
#define INL_C 41

#define IOUTA 34
#define IOUTB 35
#define IOUTC 32

LowsideCurrentSense current_sense  = LowsideCurrentSense(0.3, 25, IOUTA, IOUTB, IOUTC);

void setup() {
  // put your setup code here, to run once:
  current_sense.init();

  Serial.begin(115200);
  while(!Serial);
  Serial.println("Serial init success!");

}

void loop() { 
 
  PhaseCurrent_s currents;
  
  currents.a = analogRead(34);

    Serial.print(currents.a*1000); // milli Amps
    Serial.print("\t");
    Serial.print(currents.b*1000); // milli Amps
    Serial.print("\t");
    Serial.print(currents.c*1000); // milli Amps
    Serial.print("\t");
}

Sadudu, 2022-12-15:
Hello, Have you solved this problem? I meet the same problem.

runger1101001, 2022-12-15:
Hi,

I think you can't use the analogRead() method if you've initialised the current sensing...

But this would not be the right way to get the phase currents in any case. You would do something like this:

```
    PhaseCurrent_s currents = current_sense.getPhaseCurrents();
    float current_magnitude = current_sense.getDCCurrent();

    Serial.print(currents.a*1000); // milli Amps
    Serial.print("\t");
    Serial.print(currents.b*1000); // milli Amps
    Serial.print("\t");
    Serial.print(currents.c*1000); // milli Amps
    Serial.print("\t");
    Serial.println(current_magnitude*1000); // milli Amps
}
```

runger1101001, 2022-12-15:
If you need to use analogRead() for other purposes, please try the method:

`uint16_t value = adcRead(pin);`

instead of using analogRead().

closed 2022-12-15 by runger1101001