simplefoc/Arduino-FOC #228: Makefiles ?

weiminshen99, 2022-11-02:
For many people who do not use Arduino, are there any Makefiles available in the repository for compiling the FOC software and the related examples using "make" on a terminal ?

If not, would you like to include those Makefiles if they are developed? 

Such Makefiles could greatly increase the portability of FOC, and would be very useful for many software engineers who program in a way that is independent from any existing IDEs.

runger1101001, 2022-11-04:
Hi @weiminshen99 ,

As an Arduino library you need Arduino framework to compile SimpleFOC.

Almost all our users use either ArduinoIDE or PlatformIO to compile.

In theory, we could provide Makefiles and compile instructions for the arduino-cli, but I think it would be a large effort (we support 14 different MCU architectures), with limited returns, since hardly anyone is working in this way.

Nonetheless, if anyone gets it working and would like to provide a pull-request, we would be happy to accept it. :-)

runger1101001, 2022-11-04:
On another note, we support headless compilation on GitHub actions for our test cases.

These builds are based on docker images... I assume there would also be a way to use and run these locally, so this could be another avenue to explore...

weiminshen99, 2022-11-04:
Thanks for your nice and detailed replies, and I really appreciate them.
I will try to get the Makefile working for Bluebills first, and then let you know the results.
If you have any tips for that, please do let me know.
Have a great day!!

runger1101001, 2023-02-05:
I hope you have found a good solution for your setup! I'm going to close this for now. 

Please also see:
https://github.com/simplefoc/Arduino-FOC/issues/247

I think we will add a section to our documentation that points to instructions for using arduino-cli to build projects. I hope this will help people in future.

closed 2023-02-05 by runger1101001