simplefoc/Arduino-FOC #238: current sense: RP2040 DMA ADC wrong channel mask calculation

greymfm, 2023-01-10:
This channel mask calculation is wrong (e.g. it should compute "enableMask=7" for ADC0,1,2 but it computes 14) :
https://github.com/simplefoc/Arduino-FOC/blob/e9ac395cb9d698c0aa6d779206a6c413fbedcbb3/src/current_sense/hardware_specific/rp2040/rp2040_mcu.cpp#L149

The correct (and working) code is like this (correctly computes enableMask):
[image]

:-)

runger1101001, 2023-01-10:
Hmmm, did I get the order wrong? I just looked at the datasheet, and while the bits 20:16 define the channels to be sampled, it doesn't say which bit corresponds to what channel... :-(

The online docs for adc_set_round_robin() here
https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__adc.html#ga16d25be75e16f0671d3e3185b0b59771 

say the following: _"Value between 0 and 0x1f (bit 0 to bit 4 for GPIO 26 to 29 and temperature sensor input respectively)"_

So to be this implies that GPIO 26, which is ADC channel 0, has bit 0, while GPIO 29, ADC Channel 3 has bit 3... I don't support bit 4 (the temperature sensor).

So in my code channelsEnabled[0] == ADC Channel 0 == GPIO 26 - and this loop will set that into bit 0.

Are you saying it is the other way round?

This code is not yet very well tested. You should probably use the version from the dev branch if you aren't already, this has the most recent bug-fixes and changes...

runger1101001, 2023-01-10:
Ah, I see the problem! :-) there is one shift too many being done by this loop! I'll commit this change right away!

Done, as part of: https://github.com/simplefoc/Arduino-FOC/pull/241

So it is on the dev branch now :-)

closed 2023-03-18 by askuric