simplefoc/Arduino-FOC #239: PWM generic driver: missing code for 6PWM driver

greymfm, 2023-01-10:
The _writeDutyCycle6PWM is missing in:
https://github.com/simplefoc/Arduino-FOC/blob/dev/src/drivers/hardware_specific/generic_mcu.cpp

The missing (and tested) code is like this:

'''
// Configuring PWM frequency, resolution and alignment
// - BLDC driver - 6PWM setting
// - hardware specific
__attribute__((weak)) void* _configure6PWM(long pwm_frequency, float dead_zone, const int pinA_h, const int pinA_l,  const int pinB_h, const int pinB_l, const int pinC_h, const int pinC_l){
  GenericDriverParams* params = new GenericDriverParams {
    .pins = { pinA_h, pinA_l, pinB_h, pinB_l, pinC_h, pinC_l },    
    .pwm_frequency = pwm_frequency,
    .dead_zone = dead_zone 
  };
  return params;
}

// Function setting the duty cycle to the pwm pin (ex. analogWrite())
// - BLDC driver - 6PWM setting
// - hardware specific
__attribute__((weak)) void _writeDutyCycle6PWM(float dc_a,  float dc_b, float dc_c, PhaseState *phase_state, void* params){
    float dz = 0;
    dz = ((GenericDriverParams*)params)->dead_zone;
   analogWrite(((GenericDriverParams*)params)->pins[0], 255.0f*dc_a);
   analogWrite(((GenericDriverParams*)params)->pins[1], 255.0f*min(1.0, (dc_a + dz)));
   analogWrite(((GenericDriverParams*)params)->pins[2], 255.0f*dc_b);
   analogWrite(((GenericDriverParams*)params)->pins[3], 255.0f*min(1.0, (dc_b + dz)));
   analogWrite(((GenericDriverParams*)params)->pins[4], 255.0f*dc_c);
   analogWrite(((GenericDriverParams*)params)->pins[5], 255.0f*min(1.0, (dc_c + dz))); 
  _UNUSED(phase_state);
}
'''

askuric, 2023-01-10:
Hey @greymfm,

This is intentionally left unimplemented. The generic driver for 6pwm does not really exist and your code might work for your case but will not work in most cases. And due to the inherent complexity od 6pwm implementations on different hardware platforms we prefer to leave these functions unimplemented.

closed 2023-01-10 by askuric

greymfm, 2023-01-15:
You are absolutely right - In this moment I see the problem with my code (it only produces a deadtime at the falling edge but not at the rising edge ...) - I  was missing a comment like yours in the generic code ;-) (like 'you cannot implement a deadtime in a generic way, it is always hardware-specific...') :-)