simplefoc/Arduino-FOC #242: Add PICO SDK support for RP2040 specific driver

dacoolinus, 2023-01-10:
I am working on a platformIO project targeting the RP2040 using the WIZIO-PICO framework. The Simple-FOC library for platformIO drops in almost flawlessly with the exception of the RP2040 specific driver. The problem is the pwm_gpio_to_slice_num function. In the PICO SDK this is included with the "hardware/pwm.h" header. If I add this to the rp2040_mcu.cpp file, everything works correctly.

Could there be an additional flag added to the driver? Maybe TARGET_RP2040_BAREMETAL or something similar to include this header file?

runger1101001, 2023-01-11:
Hey, if you have a chance you could try the version of the library from the dev branch on GitHub.
Another user has recently submitted the same bug, and we have fixed it on the dev branch.

It now compiles for both the earlehillpower framework as well as the official one. It would be cool to know if it works on the WIZIO-PICO framework too...

runger1101001, 2023-03-21:
The fix has been merged in the 2.3.0 release.

closed 2023-03-21 by runger1101001