simplefoc/Arduino-FOC #246: [FEATURE] Include motor inductance in voltage ocntrol.

dzid26, 2023-01-27:
I want to just lay out the foundation for a possible improvement, and also get feedback to verify the idea.

### Issue 
Lemma 1
Current is proportional to forward torque but only when its vector is at a 90deg electrical angle.  
Lemma 2
Current always lags the rotating voltage vector. 

In the present Voltage control scheme, a 90deg electrical angle is not maintained when speed increases.
This is the present control illustration https://github.com/simplefoc/Arduino-FOC/blob/master/src/StepperMotor.cpp#L294-L298:
Vd=0
Vq=B**E**MF+ I<sub>target</sub> R
[image]
([Source](https://endless-sphere.com/forums/viewtopic.php?f=30&t=93818&p=1376865&hilit=foc+inductance#p1376865))

### Proposal
The voltage vector needs to be phase-shifted to compensate for the current vector lag. Since Park transform is already implemented, the phase lead angle can happen naturally by manipulating Vd. (White vector below)
The control should look like this:
[image]
Setting [Vd](https://github.com/simplefoc/Arduino-FOC/blob/master/src/StepperMotor.cpp#L298)=I<sub>target</sub> * ω*R<sub>L</sub> will shift the effective current target vector to the desired 90deg. Only then this current will be proportional to torque. 
ω is the electrical angular speed in rad/s. I.e. shaft speed * [poles_pairs](https://github.com/simplefoc/Arduino-FOC/blob/master/src/common/foc_utils.cpp#L53). 
In case the combined motor inductance R<sub>L</sub> is unknown, it can be set to 0.
From what I know, coil inductance can be an approximation of motor inductance, but I don't know how good. On the other hand, R<sub>L</sub>can be found experimentally presumably by optimizing the value for maximum motor power.

### References:
https://docs.simplefoc.com/voltage_torque_mode#voltage-control-with-current-estimation-and-back-emf-compensation
https://endless-sphere.com/forums/viewtopic.php?f=30&t=93818&p=1376865&hilit=foc+inductance#p1376865
https://github.com/EGG-electric-unicycle/documentation/blob/master/Shane_Colton/3phduo.pdf - page 27.

runger1101001, 2023-01-28:
Thank you for this report! It makes sense to me, but I would like to think what @askuric thinks as well.

It could be easily implemented, I think as a subclass of BLDCMotor, and if it works well eventually rolled into the main implementation. 

As you point out, the main problem for users will be obtaining the value of the motor inductance if it is not given in the datasheet...

askuric, 2023-01-29:
Very interesting @dzid26, 
I like the idea very much. 
And it goes hand-in-hand with the field weakening also. We could definitely implement this in the motor classes. 
For the sensorless control we will need the inductance value as well and this enhancement could be an easy addition trough which we could start extending our API to allow users to set it.

I'm going to test this in code and report back, if the results are as expected we could make it a part of our next release.

dzid26, 2023-01-29:
Amazing. 
Maybe what is happening is that the motor starts to "experience" the field weakening and thus reaching higher speed.  
I was thinking this could happen and that's why I was thinking the best way to find optimal L value is by maximizing peak power, that is under load. 

I would love to try it. Funny thing is that I haven't run SimpleFOC on anything yet. So far I was just admiring simplicity and the good documentation :)

runger1101001, 2023-01-29:
This is very cool.

greymfm, 2023-03-11:
Just trying to understand this brilliant idea - this is related to both stepper and BLDC motors, correct? :-)

greymfm, 2023-03-22:
I have added the changes to our code, and tested it - it works beautifully :-)

askuric, 2023-03-23:
Awesome!
Yeah the steppers have the same issue. However they do have one transformation less when we calculate the Park and Clarke, as they have only two phases. 

In the release v2.3 we did not opt to include the steppers, but we will definitely include in for the next release.

Candas1, 2023-09-08:
I think the way it works now is that the waveform is saturated by the voltage limit of the [driver](https://github.com/Candas1/Arduino-FOC/blob/bf1e490fc6ec8875814b9eea503044dca5e0149b/src/drivers/BLDCDriver6PWM.cpp#L82) resulting with [overmodulation](https://docs.simplefoc.com/foc_theory#what-happens-when-i-exceed-maximal-uq) ?

Candas1, 2023-09-08:
I don't fully understand what happens when you don't do this, but here is how it should be calculated:
[image]

And at the end of [loopfoc](https://github.com/simplefoc/Arduino-FOC/blob/05954cb9691534dd478407505d810a68f1673a5e/src/BLDCMotor.cpp#L365), it should look like this:
```
// Circle limitation - Saturation with priority on d axis
  voltage.d = min(voltage.d,voltage_limit);
  voltage.q = min(voltage.q,_sqrt(voltage_limit*voltage_limit-voltage.d*voltage.d));

  // set the phase voltage - FOC heart function :)
  setPhaseVoltage(voltage.q, voltage.d, electrical_angle);

dzid26, 2023-09-09:
It makes sense to me. I think it is easier to see in the stepper controller because it has just Parke transformation. 
https://github.com/simplefoc/Arduino-FOC/blob/05954cb9691534dd478407505d810a68f1673a5e/src/StepperMotor.cpp#L367-L368
The voltage can be exceeded when the angle is say 45 deg because cos+sin adds up to sqrt(2). 

I am just not sure why Vd should be prioritized and how one magnetizes a rotor. I think that would apply to induction motors.  Here, I think the geometrical limitation from [STM32 FOC bible](https://www.st.com/resource/en/user_manual/um1052-stm32f-pmsm-singledual-foc-sdk-v43-stmicroelectronics.pdf) (chapter 4.12.1) would be more appropriate. 
[image]

Candas1, 2023-09-09:
Actually this is in chapter 4.8 about field weakening:
[image]

Candas1, 2023-09-10:
I did an experiment.
I changed the foc_current mode and removed the q axis PI so I can directly drive the q axis voltage, so it's like a foc_voltage mode.
The d axis PI is compensating the phase lag as the lag compensation does, and I displayed what the lag_compensation would be.
[image]

I tweaked the phase__inductance to get the d voltage and lag_compensation
I found out that my phase_inductance was half what it should be.
That's maybe a way to roughly estimate the phase inductance.

dzid26, 2023-09-12:
That's a cool stack exchange answer.  

>You want to prioritize d axis to keep an optimal angle and not to sacrifice efficiency when reaching the voltage limit

I think it you start to prioritize one over another then you rotate the frame reference.  

I think it should be:
 Vd_limited = min(Vd, Vd * voltage_limit / sqrt(Vd^2 + Vq^2))
 Vq_limited = min(Vq, Vq * voltage_limit / sqrt(Vd^2 + Vq^2))
as described in stm foc book.

askuric, 2023-09-12:
Hey @Candas1 and @dzid26, 
This is a great great discussion. 
I've had the issue of Vq and Vd limiting in the back of mind from when I first implemented it :D
I would love to include this solution to SimpleFOC!

Candas1, 2023-09-13:
I am happy to help. I will test more and share.

This simple example from stackoverflow also made me understand why we need decoupling.

dzid26, 2023-12-03:
I came across cool diagram that is an extension of the vector adding visualisation from the my first post, but with field weakening (negative Id current).  
[image]

[Analytical Tool to Generate Torque-Speed Characteristics for Surface Mounted PM Machines in Constant Torque and Field Weakening Regions](https://www.researchgate.net/profile/Muhammad-Raza-Khowja/publication/338205307_Analytical_Tool_to_Generate_Torque-Speed_Characteristics_for_Surface_Mounted_PM_Machines_in_Constant_Torque_and_Field_Weakening_Regions/links/5e1464c592851c8364b60556/Analytical-Tool-to-Generate-Torque-Speed-Characteristics-for-Surface-Mounted-PM-Machines-in-Constant-Torque-and-Field-Weakening-Regions.pdf)

dzid26, 2024-05-17:
I believe this code should be improved to **prioritize BEMF** voltage when limiting to supply voltage: 
https://github.com/simplefoc/Arduino-FOC/blob/c32bdd1e52c64b109d36a93378303e0068c7926c/src/StepperMotor.cpp#L300-L304

The idea is that the remaining of the Uq would then be the actual IR voltage. From that, the actual Iq (and actual torque!) can be back-calculated.  Then the actual Iq shall be used with Iq*w*L.

Here is a simplified [c code](https://github.com/dzid26/StepperServoCAN/blob/57b3d13fd5382439dcbd31b95b59e73c0e74d387/firmware/src/BSP/motor.c#L95-L114) from my other project. 
```
U_emf_sat = clip(U_emf, -U_lim, U_lim);
U_IR_sat = clip(U_IR, -U_lim - U_emf_sat, U_lim - U_emf_sat);
U_IR_sat = clip(U_IR_sat, -U_lim, U_lim);  //limit again to cover all cases
U_q = U_IR_sat + U_emf_sat;
I_q_act = (U_IR_sat * Ohm_to_mOhm / phase_R);
U_d = (-I_q_act) * phase_L * e_rad_s / H_to_uH;
```
That's what I am using and it's good enough for me (low speed operation). It's better than limiting without the prioritization, but it doesn't solve issue brought by @Candas1 about exceeding the limit circle. The exceeded voltage can be seen on this [visualization](https://www.desmos.com/calculator/fwumxqmswa) (X is `rev/s` and Y is `Voltage`). We'd need more advance limit circle limiting to avoid that.

I mentioned [earlier](https://github.com/simplefoc/Arduino-FOC/issues/246#issuecomment-1715257809) a relatively simple circle limiting.  But as can be seen now [here](https://www.desmos.com/calculator/mfnnj8cawd) above the 10rev/s the Uq stops increasing (it's slowly decreasing for the given parameters) and there is too much Ud voltage compared to the actual Iq. This would cause premature field weakening - good for the speed, but would reduce available torque/power. I would rather leave it at as I did it in the code above with simple bemf priority.

As to prioritizing Vd, this doesn't make sense to me. If Iq drops (which it will when crossing base speed), so should the Vd. I believe, that even when limiting for the limit circle the U_emf should be prioritized to remain true to FOC theory (at least in the context of Id=0, aka below field weakening). 
Unfortunately, when prioritizing for U_emf and staying within limit circle, the calculation becomes much more difficult and self-referencing. - It has annoying square roots, so I don't think it can be solved algebraically. - It would require multiple iterations to find solution - (i.e limit the outcome, recalculate Park transform or calculate square roots, then limiting again - few times). It would be ugly. But there might be away to offload this iterative calculation offline (or after motor parameters i.e. induction, Kbemf, are identified) and store as a lookup.