simplefoc/Arduino-FOC #247: [FEATURE] GNU Makefile?

bjne, 2023-01-29:
Is it possible to build this project with a standard Makefile? I do not understand how it is currently built..

runger1101001, 2023-01-29:
Hi @bjne ,

It would be possible, but it currently isn't done this way.

The library is an Arduino library, and as such it builds with the standard Arduino tools, Arduino IDE and Arduino CLI. It's officially released in the Arduino Library Manager, so to use it all you have to do is create a new sketch and reference the library, Arduino IDE will do the rest (fetching and building the library, linking it to your code).

It also works fine in PlatformIO, an alternative to Arduino IDE based on Visual Studio Code.

Personally, I don't see the need for a Makefile in this scenario, but if you'd like to create one, we can consider including it. But its not clear to me at the moment which group of users would benefit.

bjne, 2023-01-29:
So it would benefit whoever works with standard tooling without wrapping behind a gui I guess.
What I do not understand is where the actual build info for the library is located. Is it so that some
other repo/package (arduino-ide/platformio package) holds the information about how to build simpleFOC?

runger1101001, 2023-02-05:
Well, all Arduino libraries follow a standard pattern so that Arduino IDE can compile them.

In the embedded world it is not uncommon to work with static binaries, i.e. shared libraries often don't make sense on a MCU. So Arduino IDE, by default, compiles all libraries from source for the chosen target architecture. The IDE itself (or its framework packages) embeds the logic on how libraries are included with the user code to produce the final statically linked firmware image.

PlatformIO includes support for Arduino, so it also includes layers which can find, fetch and include Arduino libraries into your project.

The secret is the standardisation provided by the Arduino library format, along with, admittedly, the comparative simplicity of most ardunio libs.

Note that Arduino Sarl goes to some effort to keep the libraries "clean", and provides a library linting tooling which your library has to pass before it will be accepted into the library manager.

Note also that in terms of building without a GUI there is an Arduino-CLI, as well as GitHub actions for CI (which we use). There's also other options for building Arduino stuff (like Sloeber) which build on standard gnu toolings.

I'm going to close this now, if that's ok, as I think we are happy with the current system in terms of our needs, and while not opposed to the idea of Makefiles in principle, I'd like it to be motivated by an actual use case that needs it. In that case I expect there will be specific requirements for how the whole process should work, and we can create specific issues around that.

closed 2023-02-05 by runger1101001