simplefoc/Arduino-FOC #248: [BUG] No such file or directory

Suresh-Subedi, 2023-02-04:
I can't build platformio project with Arduino-FOC

**Describe the bug**
```
Compiling .pio\build\genericSTM32F103C8\lib4a2\Simple FOC\drivers\hardware_specific\generic_mcu.cpp.o
In file included from .pio\libdeps\genericSTM32F103C8\Simple FOC\src\current_sense\hardware_specific\stm32\stm32f1\../../../../drivers/hardware_specific/stm32/stm32_mcu.h:3,
                 from .pio\libdeps\genericSTM32F103C8\Simple FOC\src\current_sense\hardware_specific\stm32\stm32f1\stm32f1_hal.h:9,
                 from .pio\libdeps\genericSTM32F103C8\Simple FOC\src\current_sense\hardware_specific\stm32\stm32f1\stm32f1_hal.cpp:1:
.pio\libdeps\genericSTM32F103C8\Simple FOC\src\current_sense\hardware_specific\stm32\stm32f1\../../../../drivers/hardware_specific/stm32/../../hardware_api.h:4:10: fatal error: ../common/foc_utils.h: No such file or directory
    4 | #include "../common/foc_utils.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\genericSTM32F103C8\lib4a2\Simple FOC\current_sense\hardware_specific\stm32\stm32f1\stm32f1_hal.cpp.o] Error 1
========================================= [FAILED] Took 6.56 seconds =========================================
```

**Describe the hardware setup**
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
lib_deps = https://github.com/simplefoc/Arduino-FOC.git#dev
lib_archive = false

**IDE you are using**
- Platformio extension for vs code
- VS Code

askuric/Simple FOC@^2.2.3, 
https://github.com/simplefoc/Arduino-FOC.git#dev, 
https://github.com/simplefoc/Arduino-FOC.git
all give file not found error.

For askuric/Simple FOC@^2.2.3 and https://github.com/simplefoc/Arduino-FOC.git, error is: 
```
                 from .pio\libdeps\genericSTM32F103C8\Simple FOC\src\current_sense\hardware_specific\stm32\stm32f1\stm32f1_hal.cpp:1:
.pio\libdeps\genericSTM32F103C8\Simple FOC\src\current_sense\hardware_specific\stm32\stm32f1\../../../../drivers/hardware_specific/../hardware_api.h:6:10: fatal error: ../communication/SimpleFOCDebug.h: No such file or directory
    6 | #include "../communication/SimpleFOCDebug.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\genericSTM32F103C8\lib4a2\Simple FOC\current_sense\hardware_specific\stm32\stm32f1\stm32f1_hal.cpp.o] Error 1
========================================= [FAILED] Took 5.22 seconds =========================================
```

runger1101001, 2023-02-05:
Huh, I wonder what is going on here.

STM32F103 has a test case, which passed for v2.2.3 of the library:
https://github.com/simplefoc/Arduino-FOC/actions/runs/3255184662/jobs/5344237710
(logs are expired but you can see the test compilation passed)

So I think it must be something specific to your setup... did you make some changes inside the library folder? Could it be an issue with too long file names? Something like this?

runger1101001, 2023-02-05:
Dev Branch is also compiling for us on STM32F103:
https://github.com/simplefoc/Arduino-FOC/actions/runs/4037220033/jobs/6940362742
(this time you can see the logs)

runger1101001, 2023-02-05:
Which version of ststm32 platform are you using?

runger1101001, 2023-02-05:
Is it concievable that the build was somehow mixed up?
Can you fix it with a "Clean All" or a fresh project? Or if you `git pull` the dev branch version?

The reason I ask is that between the 2.2.3 version (master branch) and the dev branch the files for the driver code have moved around, we split them into subdirectories. So this error kind of makes sense if the compilation were "mixing up" the master and dev branch versions somehow... because, for example, of cached stuff or existing .o files or something?

Suresh-Subedi, 2023-02-05:
No I didn't change anything in the library manually. I do see 14 changes in git. But they are in example files and shouldn'tmatter.
[image]
I deleted .pio\libdeps\genericSTM32F103C8\Simple FOC and restared and rebuilt, but got same errors.

Path is C:\Users\ssubedi\source\repos\Suresh-Subedi\sologear-g3-15-electric-unicycle\g3-15-custom-firmware\.pio\libdeps\genericSTM32F103C8\Simple FOC

Which version of ststm32 platform are you using?
Where can I see that?

Suresh-Subedi, 2023-02-05:
It was certainly caused by too long path. I was able to build after moving project folder to user's home (C:\Users\ssubedi\source\repos\Suresh-Subedi\) folder.

closed 2023-02-05 by Suresh-Subedi