simplefoc/Arduino-FOC #252: [BUG] Cannot compile SimpleFOC with Earle Philhower's RP2040 board drivers

costasvav, 2023-02-14:
**Describe the bug**
Attempting to compile some basic SimpleFOC code for the Pico using Earle's board drivers, but receiving errors.  Code compiles just fine with the Pico MBED board drivers.

```
C:\Users\vavourak\Documents\Arduino\BLDC_Controller\BLDC_Controller.ino: In function 'void setup()':
C:\Users\vavourak\Documents\Arduino\BLDC_Controller\BLDC_Controller.ino:39:30: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   39 |   command.add('V', doVLimit, "voltage limit");
      |                              ^~~~~~~~~~~~~~~
C:\Users\vavourak\Documents\Arduino\BLDC_Controller\BLDC_Controller.ino:40:29: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
   40 |   command.add('A', doAngle, "target angle");
      |                             ^~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:18:44: error: 'ADC_CS_START_ONCE_BITS' was not declared in this scope
   18 | alignas(32) const uint32_t trigger_value = ADC_CS_START_ONCE_BITS; // start once
      |                                            ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp: In member function 'bool RP2040ADCEngine::init()':
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:148:5: error: 'adc_init' was not declared in this scope
  148 |     adc_init();
      |     ^~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:153:13: error: 'adc_gpio_init' was not declared in this scope; did you mean 'pio_gpio_init'?
  153 |             adc_gpio_init(i+26);
      |             ^~~~~~~~~~~~~
      |             pio_gpio_init
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:158:5: error: 'adc_set_round_robin' was not declared in this scope
  158 |     adc_set_round_robin(enableMask);
      |     ^~~~~~~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:159:5: error: 'adc_fifo_setup' was not declared in this scope
  159 |     adc_fifo_setup(
      |     ^~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:169:9: error: 'adc_set_clkdiv' was not declared in this scope; did you mean 'pwm_set_clkdiv'?
  169 |         adc_set_clkdiv(0);
      |         ^~~~~~~~~~~~~~
      |         pwm_set_clkdiv
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:172:9: error: 'adc_set_clkdiv' was not declared in this scope; did you mean 'pwm_set_clkdiv'?
  172 |         adc_set_clkdiv(48000000/samples_per_second);
      |         ^~~~~~~~~~~~~~
      |         pwm_set_clkdiv
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:185:10: error: 'adc_hw' was not declared in this scope
  185 |         &adc_hw->fifo,  // source
      |          ^~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp: In member function 'void RP2040ADCEngine::start()':
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:226:13: error: 'adc_select_input' was not declared in this scope
  226 |             adc_select_input(i); // set input to first enabled channel
      |             ^~~~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:235:5: error: 'adc_run' was not declared in this scope
  235 |     adc_run(true);
      |     ^~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp: In member function 'void RP2040ADCEngine::stop()':
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:243:5: error: 'adc_run' was not declared in this scope
  243 |     adc_run(false);
      |     ^~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:247:5: error: 'adc_fifo_drain' was not declared in this scope
  247 |     adc_fifo_drain();
      |     ^~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

I remove the command code and get the following error:
```
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:18:44: error: 'ADC_CS_START_ONCE_BITS' was not declared in this scope
   18 | alignas(32) const uint32_t trigger_value = ADC_CS_START_ONCE_BITS; // start once
      |                                            ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp: In member function 'bool RP2040ADCEngine::init()':
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:148:5: error: 'adc_init' was not declared in this scope
  148 |     adc_init();
      |     ^~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:153:13: error: 'adc_gpio_init' was not declared in this scope; did you mean 'pio_gpio_init'?
  153 |             adc_gpio_init(i+26);
      |             ^~~~~~~~~~~~~
      |             pio_gpio_init
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:158:5: error: 'adc_set_round_robin' was not declared in this scope
  158 |     adc_set_round_robin(enableMask);
      |     ^~~~~~~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:159:5: error: 'adc_fifo_setup' was not declared in this scope
  159 |     adc_fifo_setup(
      |     ^~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:169:9: error: 'adc_set_clkdiv' was not declared in this scope; did you mean 'pwm_set_clkdiv'?
  169 |         adc_set_clkdiv(0);
      |         ^~~~~~~~~~~~~~
      |         pwm_set_clkdiv
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:172:9: error: 'adc_set_clkdiv' was not declared in this scope; did you mean 'pwm_set_clkdiv'?
  172 |         adc_set_clkdiv(48000000/samples_per_second);
      |         ^~~~~~~~~~~~~~
      |         pwm_set_clkdiv
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:185:10: error: 'adc_hw' was not declared in this scope
  185 |         &adc_hw->fifo,  // source
      |          ^~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp: In member function 'void RP2040ADCEngine::start()':
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:226:13: error: 'adc_select_input' was not declared in this scope
  226 |             adc_select_input(i); // set input to first enabled channel
      |             ^~~~~~~~~~~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:235:5: error: 'adc_run' was not declared in this scope
  235 |     adc_run(true);
      |     ^~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp: In member function 'void RP2040ADCEngine::stop()':
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:243:5: error: 'adc_run' was not declared in this scope
  243 |     adc_run(false);
      |     ^~~~~~~
c:\Users\vavourak\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:247:5: error: 'adc_fifo_drain' was not declared in this scope
  247 |     adc_fifo_drain();
      |     ^~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

**Describe the hardware setup**
- Which motor: GM5208-24
- Which driver: SimpleFOCmini
- Which microcontroller: Pi Pico
- Which position sensor: AS5048A
- Current sensing used? No

**IDE you are using**
- Arduino IDE

runger1101001, 2023-02-18:
Hmmm... I actually recently added the compatibilty with the earlehillpower framework to the dev branch of the library.
I tested it with PlatformIO and the Pico W board type from that framework.

I shall have to try again to test.

Are you using ArduinoIDE?

costasvav, 2023-02-18:
I'm using the Arduino IDE 2.0.3 and a Pico (non-W).

runger1101001, 2023-03-21:
The fix has been merged in the current release (2.3.0).

closed 2023-03-21 by runger1101001