simplefoc/Arduino-FOC #264: [BUG] RP2040 ADC Engine missing include for arduino-pico core

nanoparticle, 2023-03-22:
Using the Ardiuno IDE 2 with latest SimpleFOC library and latest[ Arduino-Pico core](https://github.com/earlephilhower/arduino-pico), with my [RP2040 based board](https://oshwlab.com/nekosystems/bl2040-mini-alpha-v3), and the current sense code fails to compile with the following error:

    c:\Users\...\Documents\Arduino\libraries\Simple_FOC\src\current_sense\hardware_specific\rp2040\rp2040_mcu.cpp:19:44: error: 'ADC_CS_START_ONCE_BITS' was not declared in this scope
    19 | alignas(32) const uint32_t trigger_value = ADC_CS_START_ONCE_BITS; // start once

Several more errors follow, but they are a consequence of this error and go away when it is resolved.

It looks like [rp2040_mcu.cpp](https://github.com/simplefoc/Arduino-FOC/blob/master/src/current_sense/hardware_specific/rp2040/rp2040_mcu.cpp) in the current_sense folder is missing this include statement:
`#include "hardware/adc.h"`

When I add this line everything compiles and runs successfully. There is even a drastic improvement in the smoothness of motor compared to last release of SimpleFOC, thanks so much for the development effort! 😄

runger1101001, 2023-05-12:
I looked into this again today. I have added the line you suggest.

The code compiles for me now using:
- ArduinoIDE 2 and Arduino Pico MBed Core 4.0.2
- PlatformIO and Arduino Pico Platform 1.9.0
- PlatformIO and Earlehillpower Platform 1.7.0+sha.0c33219

It's really confusing with all these different framework versions for the same MCU :-(

nanoparticle, 2023-06-06:
Is there a particular Arduino core that is supported/preferred by SimpleFOC?

runger1101001, 2023-06-06:
Well we’d like to support the “official” one but they don’t seem to do to good a job of keeping it updated :-(
So many people seem to like alternative cores.

the current dev branch code should work with both the official core and the earlephilhower one.

please let me know if you find it doesn’t work for you.

nanoparticle, 2023-06-07:
Thanks, I'll post an update in a couple of weeks when the new boards arrive and I have had a chance to check compatibility.

closed 2023-09-24 by runger1101001