simplefoc/Arduino-FOC #269: Arduino uno to control sd30m60ac for inverter motor

Sekai1942, 2023-04-29:
Hi. Is there a posibility to control a inverter motor from an external ac unit using arduino and a SD30M60AC?
The SD30M60AC is a 3 phase full bridge driver perfect for the job.
I tried this code:

```
// BLDC driver standalone example
#include <SimpleFOC.h>

// BLDC driver instance
BLDCDriver6PWM driver = BLDCDriver6PWM(5, 6, 9,10, 3, 11, 8);

void setup() {
  
  // pwm frequency to be used [Hz]
  driver.pwm_frequency = 20000;
  // power supply voltage [V]
  driver.voltage_power_supply = 12;
  // Max DC voltage allowed - default voltage_power_supply
  driver.voltage_limit = 12;
  // daad_zone [0,1] - default 0.02 - 2%
  driver.dead_zone = 0.05;

  // driver init
  driver.init();

  // enable driver
  driver.enable();

  _delay(1000);
}

void loop() {
    // setting pwm
    // phase A: 3V, phase B: 6V, phase C: 5V
    driver.setPwm(3,6,5);
}

```
but the motor does not spin. 
it just gives one pulse at the start.

askuric, 2023-04-30:
Hey @Sekai1942,

I've gone really quickly through the datasheet and it seems that it should be possible to use it with SimpleFOC.

However the code that you've posted is not intended to spin the motor. It will only set a certain fixed voltage on each of the motor phases. 

To spin the motor in the open loop use the open loop motion control. Like velocity open loop or position open loop. 
You can find examples of Arduino sketches in the library examples folder.

Sekai1942, 2023-04-30:
> Hey @Sekai1942,
>
> I've gone really quickly through the datasheet and it seems that it should be possible to use it with SimpleFOC.
>
> However the code that you've posted is not intended to spin the [...]

something like this:
```
// Open loop motor control example
#include <SimpleFOC.h>

// BLDC motor & driver instance
// BLDCMotor motor = BLDCMotor(pole pair number);
BLDCMotor motor = BLDCMotor(11);
// BLDCDriver3PWM driver = BLDCDriver3PWM(pwmA, pwmB, pwmC, Enable(optional));
//BLDCDriver3PWM driver = BLDCDriver3PWM(9, 5, 6, 8);
BLDCDriver6PWM driver = BLDCDriver6PWM(5, 6, 9,10, 3, 11);
// Stepper motor & driver instance
//StepperMotor motor = StepperMotor(50);
//StepperDriver4PWM driver = StepperDriver4PWM(9, 5, 10, 6,  8);

//target variable
float target_velocity = 0;

// instantiate the commander
Commander command = Commander(Serial);
void doTarget(char* cmd) { command.scalar(&target_velocity, cmd); }
void doLimit(char* cmd) { command.scalar(&motor.voltage_limit, cmd); }

void setup() {

  // driver config
  // power supply voltage [V]
  driver.voltage_power_supply = 12;
  // limit the maximal dc voltage the driver can set
  // as a protection measure for the low-resistance motors
  // this value is fixed on startup
  driver.voltage_limit = 6;
  driver.init();
  // link the motor and the driver
  motor.linkDriver(&driver);

  // limiting motor movements
  // limit the voltage to be set to the motor
  // start very low for high resistance motors
  // current = voltage / resistance, so try to be well under 1Amp
  motor.voltage_limit = 3;   // [V]
 
  // open loop control config
  motor.controller = MotionControlType::velocity_openloop;

  // init motor hardware
  motor.init();

  // add target command T
  command.add('T', doTarget, "target velocity");
  command.add('L', doLimit, "voltage limit");

  Serial.begin(115200);
  Serial.println("Motor ready!");
  Serial.println("Set target velocity [rad/s]");
  _delay(1000);
}

void loop() {

  // open loop velocity movement
  // using motor.voltage_limit and motor.velocity_limit
  motor.move(target_velocity);

  // user communication
  command.run();
}
```
i uploaded the code and the motor just stalls and after few seconds the resistor that was on the curent sensing just exploded.

runger1101001, 2023-11-28:
Ouch, I'm very sorry to hear this!

I am not sure exactly what has happened in your case.

Since this is now quite an old issue, if it is ok I will close it. I encourage you to ask more questions in our forum: https://community.simplefoc.com

closed 2023-11-28 by runger1101001