simplefoc/Arduino-FOC #272: Industrial servo motor

kilofeq, 2023-05-18:
Hi! I've got a servo motor 90ST-M20430. That's the AC servo motor 220v, 750w, maximum of power 750W. Is it possible to control it using simpleFOC with a small amount of speed? Actually, I'd like to have a maximum of 150rpm, so that would be a small part of maximum voltage, right? Thanks for your reply!

runger1101001, 2023-11-28:
Wow, ok, sorry, I missed this question!

So it looks to me like this AC servo motor is a two phase induction motor. You can't drive such motors with SimpleFOC. We can drive 3 phase motors and stepper motors, but not AC induction motors.

In answer to your other question, yes, SimpleFOC can control very low speeds. In order for SimpleFOC to do field-oriented-control (closed loop) you will need a position sensor on the motor. Then the efficiency will also be much better than in open loop mode.

runger1101001, 2023-11-28:
if its ok, I will now close this question issue. Please feel free to post more questions, either here, or even better on our forums: https://community.simplefoc.com

closed 2023-11-28 by runger1101001