simplefoc/Arduino-FOC #279: [BUG] Cannot disable Debug print

robcazzaro, 2023-06-14:
I'm using Platformio, and I'm trying to reduce the size of a SimpleFOC project running on a GD32F130C8 processor with only 32K FLASH.

One obvious way is to reduce the size of the debug messages. I looked at the various messages, and in the file SimpleFOCDebug.h I found this

```
 * Add -DSIMPLEFOC_DISABLE_DEBUG to your compiler flags to disable debug in
 * this way.
```

When I set that build flag, though, it fails in FOCMotor.cpp in [line 85](https://github.com/simplefoc/Arduino-FOC/blob/05954cb9691534dd478407505d810a68f1673a5e/src/common/base_classes/FOCMotor.cpp#L85), trying to enable a non-existing class 

```
void FOCMotor::useMonitoring(Print &print){
  monitor_port = &print; //operate on the address of print
  SimpleFOCDebug::enable(&print);
  SIMPLEFOC_DEBUG("MOT: Monitor enabled!");
}
```

RoboDurden, 2023-06-15:
Don't know if this will help the simpleFOC code but i always use debug macros that either map to serial ouput code or nothing:

```
#ifdef SERIALDEBUG
	#define DEBUG(code)	{code}
	#define OUT(s)	{SERIALDEBUG.print(s);}
	#define OUT2(s,i)	{SERIALDEBUG.print(s);SERIALDEBUG.print(": ");SERIALDEBUG.print(i);}
	#define OUT2T(s,i)	{SERIALDEBUG.print(s);SERIALDEBUG.print(": ");SERIALDEBUG.print(i);SERIALDEBUG.print("\t");}
	#define OUTN(s)	{SERIALDEBUG.println(s);}
#else
	#define DEBUG(code)
	#define OUT(s)
	#define OUT2(s)
	#define OUT2T(s)
	#define OUTN(s)
#endif
```

usage:

```
  DEBUG( 
    OUT2T("GD32",iNow) 
    for (int i=0; i<HALL_Count; i++)  OUT2T(i,aoHall[i].Get())
    OUT2T("angle",sensor.getAngle()) 
    OUT2T("speed",sensor.getVelocity())
    OUT2T("iI2cReq",iI2cReq);
    OUT2("iI2cRec",iI2cRec);
    OUTLN()
  )
```

runger1101001, 2023-09-02:
Fixed this on the dev branch @robcazzaro , thanks for reporting it!

PR #303 (2023-09-02): fix errors and warnings on SIMPLEFOC_DISABLE_DEBUG
Fix #279
closed 2023-09-24 by runger1101001