simplefoc/Arduino-FOC #282: [FEATURE] TLE5012B MagneticSensorSPI

huigang39, 2023-06-19:
**Is your feature request related to this issue? Please describe. **
How to add support for new sensors?

**Describe your desired solution**
I don't know how to add TLE5012B MagneticSensor.

runger1101001, 2023-06-19:
We have a driver for this sensor for STM32 MCUs here: https://github.com/simplefoc/Arduino-FOC-drivers/tree/master/src/encoders/tle5012b

It is part of the SimpleFOCDrivers library, which is a seperate library from the main one.

There are also some instructions for using it with other MCU types.

runger1101001, 2023-09-24:
If it's ok, I'm going to close this issue now. If you have more questions about using the TLE5012B with SimpleFOC, please use our forums at community.simplefoc.com

closed 2023-09-24 by runger1101001