simplefoc/Arduino-FOC #292: [ESP32S2, ARDUINO] MOTOR SKIPS EVERY 2 SEC, EVEN IN OPEN LOOP[BUG]

rm2488, 2023-07-17:
**Describe the bug**
The motor skips approx every 2 seconds, in open loop as well as closed loop. 

**Describe the hardware setup**
- Which motor - 1.8 degreee stepper
- Which driver - Custom 4PWM driver
- Which microcontroller - ESP32S2
- Which position sensor - Irrelevant
- Current sensing used? - NO

- Arduino IDE 

While I am submitting a bug report, I already have a solution, so please forgive me if this is in the wrong place. There may be better solutions out there.

I investigated everything (Including a lot of time looking at the total angle stored as float precision issue) until it hit me- 

this code has the glitch- (paraphrasing for clarity)

**void setup()
{
do_setup();
}

void loop()
{
motor.loopFOC();
motor.move();
motor.monitor();
command.run();
}**

This one fixes it:

**void setup()
{
do_setup();
while(1)
  {
  loop();
  };
}

void loop()
{
motor.loopFOC();
motor.move();
motor.monitor();
command.run();
}**

Effectively equivalent to placing the contents of the loop function in the while loop. Therefore, the loop function never runs as a task in FreeRTOS. Otherwise there is apparently some delay between loops from time to time.

This issue is explored here:
https://www.reddit.com/r/esp32/comments/t9aqr6/what_on_earth_is_my_esp32_doing_between_loops/

It may be specific to the ESP32S2, and there may be other repercussions to my workaround.

runger1101001, 2023-09-03:
Hey, thank you for reporting this.

May I ask, are you using any of the chips radio functions like BLE or WiFi in your code? Or is it just SimpleFOC with no other functions?

It's certainly an interesting finding, and important for us as many people do use ESP32. I wonder why it has not been reported before? Can you think of something unique in your setup that could be causing it?

askuric, 2025-07-29:
I'm gonna close this issue. Thanks for reporting it, it will surely help. People experiencing the same behavior.

closed 2025-07-29 by askuric