simplefoc/Arduino-FOC #293: [BUG]simple dc_open_loop example not working with esp32 and L296n

disc0018, 2023-07-17:
I tried to use I simple esp32 and L296n for test reasons. 
I was not able to get a working PWM signal.
I changed the DCDriver to DCDriver1PWM2Dir driver = DCDriver1PWM2Dir(14, 2, 27);
(Hardwareconfig was ok. A simple ledcWrite(ledChannel, dutyCycle) was working for test.)

It seems like the following hardware specific code in esp32_ledc_mcu.cpp is never reached:
......

```
void _writeDutyCycle1PWM(float dc_a, void* params){
  ledcWrite(((ESP32LEDCDriverParams*)params)->channels[0], _constrain(_PWM_RES*dc_a, 0, _PWM_RES));
}
```

runger1101001, 2023-07-18:
Which ESP32 are you using?

The LEDC driver is only used on some ESP32 models, the ones that don’t have MCPWM peripherals…
To force the use of the LEDC driver, you can set the build flag:
-DSIMPLEFOC_ESP32_USELEDC

unfortunately, we don’t yet support 1-PWM on ESP32 with MCPWM, you have to use LEDC.

the DCDrivers and 1-PWM mode are quite new, you might be the first person testing it on ESP32. Please forgive any bugs you may encounter.

disc0018, 2023-07-18:
Hello, I am using an old esp32 (ESP32-D0WDQ6). I try to use the build
flag...

disc0018, 2023-07-18:
I tested it,

and came to the following results:
The generic_mcu part is used in every case
and analogWrite does not produce an output...
If I add
...
ledcSetup(0, 5000, 8);
ledcAttachPin(14, 0);
...
to my setup() the generic_mcu works.
the esp analogWrite is used and I find a PWM Signal on pin 14...
https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/ledc.html

If I force your code to use the esp32_ledc_mcu.cpp
it works without ledcSetup and ledcAttachPin in my setup().

the if defined ....
in generic_mcu.cpp / esp32_ledc_mcu.cpp is not working co.

runger1101001, 2023-07-20:
Hey,

that’s strange. Maybe we are missing some defines to handle this MCU?

normally the code should always prefer to use either LEDC or MCPWM on a MCU it recognizes.

are you using PlatformIO? If so, do you have the option `lib_archive = false` in your platformio.ini? That would be one explanation

disc0018, 2023-07-21:
Hello,
I am using the Arduino_IDE 2.1.1
and the newest libraries.

I found esp32_ledc_mcu.cpp is not used, because SOC_MCPWM_SUPPORTED is
defined
......
#if defined(ESP_H) && defined(ARDUINO_ARCH_ESP32) && (
!defined(SOC_MCPWM_SUPPORTED) || defined(SIMPLEFOC_ESP32_USELEDC) )
......

esp32_driver_mcpwm.h/esp32_mcu.cpp is not used because there is no void*
_configure1PWM/void _writeDutyCycle1PWM
implemented, so it falls back to generic_mcu.cpp.
Here....
#if defined(ESP_H) && defined(ARDUINO_ARCH_ESP32)
  __attribute__((weak)) void analogWrite(uint8_t pin, int value){ };
#endif
is linked to analogWrite in the esp32 implementation.

but because of a bug?.... in the esp32 analogWrite implementation the
following code fails:
_attribute__((weak)) void _writeDutyCycle1PWM(float dc_a, void* params){
  // transform duty cycle from [0,1] to [0,255]
  analogWrite(((GenericDriverParams*)params)->pins[0], 255.0f*dc_a);
}
because the esp32 needs before analogWrite the call of
ledcSetup and ledcAttachPin.... (fully implemented in esp32_ledc_mcu.cpp)

the use of the esp32_ledc_mcu.cpp with
...define SIMPLEFOC_ESP32_USELEDC 1... fails. I do not understand why.
The code inside...
#if defined(ESP_H) && defined(ARDUINO_ARCH_ESP32) && (
!defined(SOC_MCPWM_SUPPORTED) || defined(SIMPLEFOC_ESP32_USELEDC) )
... is never used.

The best solution? I do not know.
Is there an reason for the missing implementation of
void* _configure1PWM/void _writeDutyCycle1PWM in esp32_mcu.cpp?

The SIMPLEFOC_ESP32_USELEDC flag is not working for me.
But I do not understand the problem, it should work... but it does not.

Perhaps you have an idea why the flag fails?

runger1101001, 2023-09-02:
Where are you putting this flag, SIMPLEFOC_ESP32_USELEDC ?

it has to go in the board definition or platform files to make sure the library is compiled with it…

disc0018, 2023-09-03:
Hello,
I am not sure. It was just for testing.
I was using the Arduino IDE and changed the source files directly.
If I have tried the board definitions .... I am not sure.
In the end I created in esp32_mcu.cpp the missing functions

*_configure1PWM....*
*_writeDutyCycle1PWM....*

witch I simply linked to

_configure2PWM
_writeDutyCycle2PWM

(wasting MCPWM_TIMER_1....)

and my test code was working, using the MCPWM of the esp32 as intended....

Dirk

runger1101001, 2023-09-05:
Ok,understood, thank you!

We'll leave this open until we've added the 1PWM functions to the MCPWM driver.