simplefoc/Arduino-FOC #294: [FEATURE] Support 6-channel PWM with only TC4 on ATmega32u4

cinderblock, 2023-07-19:
I'd like to use some existing ATmega32u4 based driver boards. Theses are wired up to the 6-channel PWM output that is unique to the [ATmega32u4](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf)'s `10-bit High Speed Timer/Counter4`.

| AVR Pin | Output Compare | Arduino Pin |
| ------- | -------------- | ----------- |
| PC7     | OC4A           | D13         |
| PC6     | OC4A~          | D5          |
| PB6     | OC4B           | D10         |
| PB5     | OC4B~          | D9          |
| PD7     | OC4D           | D6          |
| PD6     | OC4D~          | D12         |

Looking at the sources in `atmega32u4_mcu.cpp`, it's clear that the code is only set up for a couple dedicated pins that don't fully overlap with the Timer 4 hardware outputs.

I realize this is an older IC but is there any interest in supporting this mode?

My very old implementation: https://github.com/cinderblock/3-Phase-Controller/blob/master/src/ThreePhase/Driver.cpp

----

PS, great to meet you at OpenSauce!

cinderblock, 2023-07-19:
Yes, I'm interested. But I won't have time to look much at this for the next 2, maybe 6+ weeks! (Moving offices, Burning Man, traveling to a wedding) So if anyone else wants to get started, be my guest!

---

To be sure, would this (new) code live in this repo or in the drivers repo?

runger1101001, 2023-07-20:
From my point of view it would be one of the PWM drivers in the main repository, or a modification to one of the existing AVR drivers…