simplefoc/Arduino-FOC #301: [FEATURE] Arduino Nano ESP32 support

dderg, 2023-09-01:
Arduino nano ESP32 has ESP32-S3 chip, so it seems like it should be supported, but I get the following error, looks like the library doesn't know what board it is:

```
In file included from /Users/daniladergachev/Library/Arduino15/packages/arduino/hardware/esp32/2.0.11/cores/esp32/Arduino.h:223,
                 from /Users/daniladergachev/Documents/Arduino/libraries/Simple_FOC/src/drivers/hardware_specific/../../common/foc_utils.h:4,
                 from /Users/daniladergachev/Documents/Arduino/libraries/Simple_FOC/src/drivers/hardware_specific/../hardware_api.h:4,
                 from /Users/daniladergachev/Documents/Arduino/libraries/Simple_FOC/src/drivers/hardware_specific/generic_mcu 3.cpp:1:
/Users/daniladergachev/Library/Arduino15/packages/arduino/hardware/esp32/2.0.11/cores/esp32/io_pin_remap.h:25:49: error: variable or field 'analogWrite' declared void
 #define analogWrite(pin, value)     analogWrite(digitalPinToGPIONumber(pin), value)
```

dderg, 2023-09-01:
Seems like this code is executed, but not necessary for this board:

```
#if defined(ESP_H) && defined(ARDUINO_ARCH_ESP32)
  __attribute__((weak)) void analogWrite(uint8_t pin, int value){ };
#endif
```

runger1101001, 2023-09-02:
I have run SimpleFOC on the ESP32 Nano with the latest framework versions.

be aware that when using the Nano ESP32 it uses Arduino PIN numbers, which is unusual in my experience, normally you work with the ESP32 gpio numbers.

the SimpleFOC code has a bug with respect to this, meaning you have to use the gpio numbers of the pins when initializing BLDCDriver.

i have not yet tested current sensing.

to get rid of the compile error please use the dev branch of SimpleFOC or wait for the next release…

runger1101001, 2023-09-02:
I had fixed it like this:

```
// if the mcu doen't have defiend analogWrite
#if defined(ESP_H) && defined(ARDUINO_ARCH_ESP32) && !defined(analogWrite)
  __attribute__((weak)) void analogWrite(uint8_t pin, int value){ };
#endif
```

runger1101001, 2023-09-02:
I have committed this to the dev branch of SimpleFOC, please let me know if it works for you.

dderg, 2023-09-06:
I think you can still use GPIO pins, you just need to define them as GPIOx. I will give it a try, thank you!

dderg, 2023-09-07:
It works, but for interrupts to work I had to remove digitalPinToInterrupt, it was returning an integer and then attachInterrupt was confusing it with another pin number, at least that's what I think was happening.

closed 2023-09-24 by runger1101001