simplefoc/Arduino-FOC #306: [BUG] POW()

Candas1, 2023-09-06:
Hi,

A user came across something weird when testing on a gd32 board.
A change in one of my last commit is making the main loop much slower, impacting motor control.

It seems the reason for the performance drop is the usage of pow() in the [init of the voltage sense](https://github.com/simplefoc/Arduino-FOC-drivers/blob/7eb7bbeda90fbb024f7a02227b373ea8518065d7/src/voltage/GenericVoltageSense.cpp#L24), although it's not even running  in the main loop.... I still need to figure out why this happens, the gd32 is using a [4 years old gcc](https://registry.platformio.org/tools/platformio/toolchain-gccarmnoneeabi?version=1.90201.191206).

On stm32, I haven't noticed the performance drop, but I see another impact, which is the flash usage.
This is without the voltage sense:
```
RAM:   [=         ]   9.2% (used 4520 bytes from 49152 bytes)
Flash: [===       ]  27.3% (used 71484 bytes from 262144 bytes)
```

This is with volage sense:
```
RAM:   [=         ]   9.3% (used 4572 bytes from 49152 bytes)
Flash: [===       ]  29.5% (used 77332 bytes from 262144 bytes)
```

Now, as we are using the pow function only for powers of 2 and integers, I use this macro `#define pwrtwo(x) (1 << (x))`:
```
RAM:   [=         ]   9.3% (used 4572 bytes from 49152 bytes)
Flash: [===       ]  28.0% (used 73420 bytes from 262144 bytes)
```

So 1.5% less on a big chip (256Kb).

It seems pow() is used for the same purpose in [MagneticSensor](https://github.com/search?q=repo%3Asimplefoc%2FArduino-FOC%20pow(&type=code)

runger1101001, 2023-09-07:
Thank you for reporting this! We’ll try to fix it for next release, since using pow() seems quite gratuitous for raising to powers of 2

Candas1, 2023-09-07:
Please let me know if a PR helps
I can add `#define _powtwo(x) (1 << (x))` [here](https://github.com/simplefoc/Arduino-FOC/blob/05954cb9691534dd478407505d810a68f1673a5e/src/common/foc_utils.h#L15), update magnetic sensors init, and update arduino-foc-drivers when the first PR is accepted

askuric, 2023-09-08:
Very interesting!

Candas1, 2023-09-09:
I also checked, the svpwm without atan2 I mentioned in the community also reduces flash by 0.5%. 

You guys seem very busy, please let me know how I can make your life easier with this.

Do you prefer an issue on github for any improvement to better track. I can even create a PR.

runger1101001, 2023-09-10:
Thanks so much @Candas1 !  I'm sorry we're so busy, but don't think all your work has gone unnoticed! We're really very grateful for your support and careful testing.

I will work in the pow() changes right now. 

For the SVPWM a PR would be greatly appreciated, and also it would credit you for the change on github, which seems more than fair!! Note that we take PRs against the dev branch only.

Candas1, 2023-09-11:
Don't get me wrong, it's not a complain.

For the SVPWM change I have some questions but we can discuss it separately.

runger1101001, 2023-09-11:
pow stuff is part of PR #304

Candas1, 2023-09-11:
Thanks.
It works, I see the flash reducing, and the motor control works as usual.
I cannot test the magnetic encoder part.

closed 2023-09-24 by runger1101001