simplefoc/Arduino-FOC #307: SPI Magnetic sensor reads value correctly only once, after spi begin

RaphZufferey, 2023-09-07:
Hi! Thank you for the fantastic library, it is proving to be of great help!
I am running into an issue trying to read an SPI sensor. Sensor gets declared as instructed:

```
  MagneticSensorSPI sensor = MagneticSensorSPI(PA15, 14, 0x3FFF);
  SPIClass SPI_3(PB5, PB4, PB3); //(mosi, miso, sclk)
```

Then initialized (`sensor.init(&SPI_3);`) in the setup loop. 
I am reading the sensor in the main loop just like in the example: 
`  Serial.print((float)sensor.getSensorAngle()); Serial.print("\t");`

The issue: the sensor returns the correct value only once. On subsequent calls, it returns the same value as before. 
I managed to circumvent this issue by adding `spi->begin();` in the library, which gets then called every time:
```
// function reading the raw counter of the magnetic sensor
int MagneticSensorSPI::getRawCount(){
	spi->begin();
	return (int)MagneticSensorSPI::read(angle_register);
}
```

This is obviously not the correct way of solving this issue, and I was hoping someone could look into this?

- G431B-ESC1
- SPI sensor AS5048a connected on SPI3, thanks to the [forum](https://community.simplefoc.com/t/b-g431b-esc1-beginner-guide-i2c-guide/515/91?u=raphzufferey)
- Programmed with VSCode and PlatformIO

runger1101001, 2023-09-07:
Are you calling sensor.update() during your test?
This is called automatically by motor.loopFOC() but if you’re just testing the sensor stand-alone you have to call it yourself.

RaphZufferey, 2023-09-07:
Solved, thank you very much!

closed 2023-09-07 by RaphZufferey