simplefoc/Arduino-FOC #311: [FEATURE] Foc calculation refactoring

Candas1, 2023-09-19:
Hi,

This is not adding a feature, but could make further changes easier.
There is duplicate code [here](https://github.com/simplefoc/Arduino-FOC/blob/05954cb9691534dd478407505d810a68f1673a5e/src/common/base_classes/CurrentSense.cpp#L14) and here

``` // calculate clarke transform
    float i_alpha, i_beta;
    if(!current.c){
        // if only two measured currents
        i_alpha = current.a;  
        i_beta = _1_SQRT3 * current.a + _2_SQRT3 * current.b;
    }if(!current.a){
        // if only two measured currents
        float a = -current.c - current.b;
        i_alpha = a;  
        i_beta = _1_SQRT3 * a + _2_SQRT3 * current.b;
    }if(!current.b){
        // if only two measured currents
        float b = -current.a - current.c;
        i_alpha = current.a;  
        i_beta = _1_SQRT3 * current.a + _2_SQRT3 * b;
    } else {
        // signal filtering using identity a + b + c = 0. Assumes measurement error is normally distributed.
        float mid = (1.f/3) * (current.a + current.b + current.c);
        float a = current.a - mid;
        float b = current.b - mid;
        i_alpha = a;
        i_beta = _1_SQRT3 * a + _2_SQRT3 * b;
    }

This could be made a function getAlphaBeta or getClarkTransform that uses foc currents as inputs, and returns alpha and beta.
A function getParkTransform could use the electrical angle, alpha and beta and return DQcurrents.
getDCCurrent could use the electrical angle, alpha and beta and return DCcurrent.

Eventually, foc_current torque control type would do:
- getPhaseCurrents
- getClarkTransform
- getParkTransform
- PI control

And dc_current torque control type would do:
- getPhaseCurrents
- getClarkTransform
- getDCcurrent
- PI control

Candas1, 2023-09-19:
If this generates too much work (changing the examples, ....)
This alternative should have less impact:

getfoccurrents:
- getPhaseCurrents
- getClarkTransform
- getParkTransform

getdccurrents:
- getPhaseCurrents
- getClarkTransform
- calculate dc current

and no changes in loopfoc.

Candas1, 2023-09-24:
I only added a getABCurrents function and it looks like [that](https://github.com/Candas1/Arduino-FOC/commit/87279080e2b2d5aa25149e90df65775244eb5204)

before:
RAM:   [=         ]   9.2% (used 4536 bytes from 49152 bytes)
Flash: [===       ]  27.3% (used 71540 bytes from 262144 bytes)
loopfoc = 260us

RAM:   [=         ]   9.2% (used 4536 bytes from 49152 bytes)
Flash: [===       ]  27.3% (used 71500 bytes from 262144 bytes)
loopfoc = 260us

So no real space saving, no performance impact, but a single function to enhance in the future.
If you think it's useful I can send you the PR.

Candas1, 2023-09-24:
I thought a getDQCurrents function (park transform) wouldn't be useful because it's used only once.
But if you guys want to implement FOC for steppers, you could reuse such a function.
[Inverse park](https://github.com/simplefoc/Arduino-FOC/blob/f9e9a2d29e912957bd75302d4f9a83a1cdff37a5/src/StepperMotor.cpp#L368C18-L368C18) is used for both bldc and stepper motors also.

askuric, 2024-06-07:
@Candas1 should be close this one?

Candas1, 2024-06-09:
Sorry I forgot about this one, it was covered [here](https://github.com/simplefoc/Arduino-FOC/pull/340).

closed 2024-06-09 by Candas1