simplefoc/Arduino-FOC #320: [BUG]ESP32-S3 block in InlineCurrentSense init

mailonghua, 2023-09-27:
**Describe the bug**
When using SimpleFOC, the program will stay at the current_sense.init(); function location

**Describe the hardware setup**
For us it is very important to know what is the hardware setup you're using in order to be able to help more directly
- Which microcontroller
ESP32S3FN8
- Current sensing used?
INA240A2PWR
**IDE you are using**
- Arduino IDE 
- Platformio

**Config file**
[env:esp32-s3-devkitc-1]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
lib_deps = 
	askuric/Simple FOC @ ^2.3.1
	SPI
	Wire
lib_archive = false

**CurrentCode**
#include <SimpleFOC.h>

#define A0 8
#define A1 9
// current sensor
// shunt resistor value
// gain value
// pins phase A,B, (C optional)
InlineCurrentSense current_sense = InlineCurrentSense(0.01f, 50.0f, 8, 9);

void setup() {
  Serial.begin(115200);
  Serial.println("Start init current sense");
  // initialise the current sensing
  current_sense.init();
  Serial.println("End init current sense");

  // for SimpleFOCShield v2.01/v2.0.2
  current_sense.gain_b *= -1;
  

  Serial.println("Current sense ready.");
}

void loop() {

    PhaseCurrent_s currents = current_sense.getPhaseCurrents();
    float current_magnitude = current_sense.getDCCurrent();

    Serial.print(currents.a*1000); // milli Amps
    Serial.print("\t");
    Serial.print(currents.b*1000); // milli Amps
    Serial.print("\t");
    Serial.print(currents.c*1000); // milli Amps
    Serial.print("\t");
    Serial.println(current_magnitude*1000); // milli Amps
}

**Out**
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
Start init current sense

mailonghua, 2023-09-27:
I think I have located this problem. This is a bug. Each ADC of esp32s3 has 10 channels 0~9, and the SENS_SAR_MEAS1_CTRL2_REG register shows the status of ADC1, and SENS_SAR_MEAS2_CTRL2_REG shows the status of ADC2. But when I use GPIO9, currently The judgment detects BIT16 of SENS_SAR_MEAS2_CTRL2_REG, which belongs to ADC2. When I change if(channel > 7) to if(channel > 9), my current program is normal.
But I think this method may not be appropriate for ADC2

mailonghua, 2023-09-27:
> I think I have located this problem. This is a bug. Each ADC of esp32s3 has 10 channels 0~9, and the SENS_SAR_MEAS1_CTRL2_REG register shows the status of ADC1, and SENS_SAR_MEAS2_CTRL2_REG shows [...]
Simple FOC\src\current_sense\hardware_specific\esp32\esp32s_adc_driver.cpp :237

runger1101001, 2023-09-27:
Thank you very much for reporting this and finding the solution! We'll look into it as soon as we can!

runger1101001, 2023-09-27:
Probably a duplicate: #295

askuric, 2024-07-22:
The fix is in the new release v2.3.4

closed 2024-07-22 by askuric