simplefoc/Arduino-FOC #323: problem to notice the movement in the alignment of the sensor

AndSan97, 2023-10-02:
I'm testing the torque control algorythm with arduino UNO, simple FOC shield (version 2.0.4) and BLDC motor (8 poles) with Hall sensors integrated.
When I launch the program obtain "Failed to notice movement",what could be the problem?

Candas1, 2023-10-02:
During the sensor align procedure, SimpleFOC measures the direction of the sensor and the zero electrical angle offset.
If your motor is not moving at all, it will fail.

Is your motor not moving ?
Your driver could be set in a wrong way, or motor.voltage_sensor_align could be too low.
Is you motor moving ?
Then your sensor might be set in a wrong way.

AndSan97, 2023-10-02:
When I launch the program the motor rotates before in a direction then change the direction in order to find the zero but It fails.

Candas1, 2023-10-02:
Well you have your answer then ;)

AndSan97, 2023-10-05:
Thank you for the advices, I have tried to control the settings of Hall's sensor but obtain always the same problem.
Maybe It's related to the low number of poles?

askuric, 2026-02-21:
I'm gonna close this issue because it's becoming a bit old. Feel free to open it if you still encounter issue.