simplefoc/Arduino-FOC #325: [FEATURE] Calculate Ts for Loopfoc and move only once

Candas1, 2023-10-15:
While working on some code I had an idea for performance improvement.

Now Loopfoc and move are running in the main lopp and timestamps are used to calculate Ts:
float Ts = (timestamp_now - timestamp_prev) * 1e-6f;

If I take the worst example being foc_current, Ts is being calculated separately for:
- SmoothingSensor::update()
- LPF_current_q()
- LPF_current_d()
- PID_current_q()
- PID_current_d()
- getVelocity()
- LPF_velocity() 

Maybe making this a parameter of those functions and not calculating Ts if it's already provided would save some resources.
This is with the assumptions that an interrupt running in the middle will impact all those different Ts in the same way.

This could also be useful if someone runs Loopfoc from a timer interrupt, then the timer period could be provided to the functions.

Candas1, 2023-10-15:
[Here](https://github.com/Candas1/Arduino-FOC/commit/cceec5a117b4c913b42bd71f9264cc415dff9148) is an example after changing LPF and PID only.

I tested this on a STM32F1 in foc_current mode.

Before:
RAM:   [=         ]   9.0% (used 4432 bytes from 49152 bytes)
Flash: [===       ]  26.5% (used 69560 bytes from 262144 bytes)
loopfoc = 292us

After:
RAM:   [=         ]   9.0% (used 4440 bytes from 49152 bytes)
Flash: [===       ]  26.6% (used 69656 bytes from 262144 bytes)
loopfoc = 284us

Not a very big improvement but it can be useful.

dekutree64, 2023-10-31:
I think it should be named dt, not Ts. Timestamp usually refers to an absolute time value when an event occurred, whereas this is the change in time since last frame.

motor.move will need a different dt when downsampling is used.

I do agree factoring it into a reusable class would be nice. Then you could cleanly use function overloading rather than a default argument, like float operator() (float error) { (*this)(error, dt_calculator.update()); } so the decision on how to acquire dt is made at compile time rather than having to check for 0 every frame.

Candas1, 2023-10-31:
> I think it should be named dt, not Ts. Timestamp usually refers to an absolute time value when an event occurred, whereas this is the change in time since last frame.

True, my mistake.

> I do agree factoring it into a reusable class would be nice. Then you could cleanly use function overloading rather than a default argument, like float operator() (float error) { (*this)(error, [...]

Thank you, I am new to C++/Object oriented programming so I appreciate the ideas.

runger1101001, 2023-12-02:
I would propose another way of doing it:

At the moment we have a "sensor" abstraction for the motor, which allows to sample the rotor angle. The idea behind it is that you choose a moment to sample it, and it then holds the sampled value for anyone who needs it during the subsequent computations.

Can we do the same thing for the dt value? We choose a point to sample the clock, and then keep that sample for use in subsequent computations?

This seems easier to me than passing the value around all the functions, instead of calling _micros() we instead use something like: `motor.iteration_dt`. That value gets set at the beginning of loopFOC(), for example.

Would there be a disadvantage to using the same timestamp for all the calculations? The many values used right now can't be considered accurate, since the us accuracy and indeed the MCU clock itself aren't that accurate, and the dt value assigned to the sample during computation is close, but not exactly matched to the sensor or current sampling moment.
In fact, its kind of odd that the LPF keeps its own timestamp to begin with, since the value we want is the one associated with the sample time of the value being smoothed, and not the point in time at which we do the smoothing operation...

What do you think of `motor.iteration_dt`?

dekutree64, 2023-12-03:
That sounds the same as what we're talking about. The end users of the dt value are the lowpass filters and PIDs, which are not members of motor, so there's no getting around passing it as an argument all over the place. But it would first be calculated in loopFOC and cached in a motor member variable.

Factoring the dt calculation out into a separate class would allow the PID and lowpass classes to have a member dt_calculator which is used if no argument is given, and left to waste a bit of RAM when the argument is used. So they would retain their current functionality and add the new option for a user-specified dt, rather than making them require the dt argument. Though if all the SimpleFOC code uses the argument, then maybe it would be better to remove the old functionality so it's more obvious how they're intended to be used.

But as I said before, downsampling of motor.move will require a separate dt. Creating a second instance of dt_calculator would take care of that, whereas keeping it as inline code would likely result in one duplication even if PID and lowpass do require the argument.

Candas1, 2024-01-28:
One ideat I have is that clarke/park/filters/pids could also be in the move function, so that could also be downsampled on slow hardware.
[image]

So I think most of the need for dt would be in the move function:
LPF_current_q()
LPF_current_d()
PID_current_q()
PID_current_d()
getVelocity()
LPF_velocity()

Only the smoothing would be in loopfoc because it's needed for the position.

Copper280z, 2024-04-27:
Here are api compatible implementations for the PID controller and Low Pass filters. I break the calculation into a separate function, then use operator overloading on the call operator to handle calculating dt or using the passed in value.
https://github.com/Copper280z/Arduino-FOC/blob/dual_motor/src/common/pid.cpp
https://github.com/Copper280z/Arduino-FOC/blob/dual_motor/src/common/lowpass_filter.cpp