simplefoc/Arduino-FOC #357: [FEATURE] Add library.json to set libArchive?

timohueser, 2023-12-16:
Hi,

When using simpleFOC with PlatformIO one currently has to set `lib_archive = false` in the platformio.ini file. This is a little bug-prone, as forgetting that line leads to some hard to troubleshoot failure modes (On my RP2040 it made it impossible to change the PWM frequency, with everything else working as expected).

In my opinion a less bug-prone way of dealing with this would be to add the following **library.json** file in the root of the project: 

    {
      "build": {
        "libArchive": false
      }
    }

As far as I can tell this has the same effect and can't be forgotten by the user :)

runger1101001, 2023-12-17:
Wow, thank you for this hint! Is this a platformIO specific file or will this then also affect ArduinoIDE?

runger1101001, 2023-12-17:
Thanks for this, it looks like a PlatformIO specific file, so this looks kind of perfect for us. I will add it in the next release.

closed 2024-04-23 by askuric