simplefoc/Arduino-FOC #364: [BUG]

Kurburis, 2024-01-04:
**Bug description**
When I am using the BLDCDriver6PWM function, I get nothing on my PWM output pins. It seems that the compiler is using the _writeDutyCycle6PWM function from generic_mcu.cpp instead of the hardware-specific variants.

**Hardware**
I am using a STM32 Nucleo F303RE board.

**IDE**
- Platformio

Candas1, 2024-01-04:
Hi,

You should add a title to the issue.
Have you used this option in platformio.ini ?
`libarchive = false`

Kurburis, 2024-01-04:
> Hi,
> 
> You should add a title to the issue. Have you used this option in platformio.ini ? `libarchive = false`

Sorry for the mistake; I've added a title. 
I tried your suggestion, and it made no difference, unfortunately.

Candas1, 2024-01-04:
How are you defining the board in platformio.ini?

Kurburis, 2024-01-04:
> How are you defining the board in platformio.ini?

This is what I have so far
`[env:nucleo_f303re]
platform = ststm32
board = nucleo_f303re
framework = arduino
libarchive = false
lib_deps = askuric/Simple FOC@^2.3.2
            Wire
            SPI`

runger1101001, 2024-01-04:
It's `lib_archive` - with an underscore...

Candas1, 2024-01-04:
Oups 😂

Kurburis, 2024-01-04:
> It's `lib_archive` - with an underscore...

Tried it with the underscore, didn't change anything.

runger1101001, 2024-01-04:
The `lib_archive = false` is definately needed. If you do a make clean, it will solve the issue of not finding the correct PWM driver.

However there may be further issues with the 6PWM configuration you're using.

Please also add the following build flag to platformio.ini:

```
build_flags = -DSIMPLEFOC_STM32_DEBUG
```

And the following code to the start of your setup:

```
  Serial.begin(115200);
  SimpleFOCDebug::enable(&Serial);
  delay(5000);
```

Please let us know the output produced by the init functions.

Kurburis, 2024-01-06:
Hi, 

> The `lib_archive = false` is definately needed. If you do a make clean, it will solve the issue of not finding the correct PWM driver.
>
> However there may be further issues with the 6PWM [...]

Hi, 

I've been trying to do what you said, but I have a different problem now. Nothing I write via Serial or SerialUSB is visible on the serial monitor. I've tried googling the problem and I've enabled some of the flags, but still no success.

```
#include <Arduino.h>
void setup() {

  SerialUSB.begin(9600);
  delay(5000);
}

void loop()
{
  SerialUSB.println("Test to see if Serial works");
  delay(1000);
}

```

My ini file looks now like this:

```
[env:nucleo_f303re]
platform = ststm32
board = nucleo_f303re
framework = arduino
lib_archive = false
monitor_dtr = 0
monitor_rts = 0
build_flags = 
            -DSIMPLEFOC_STM32_DEBUG
            -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
            -D USBCON
monitor_speed = 9600
monitor_port = COM8
lib_deps = askuric/Simple FOC@^2.3.2
            Wire
            SPI
```

runger1101001, 2024-01-08:
If it's a Nucleo64 board then I think it does not have USB serial. The USB is connected to the on-board ST-Link.

The ST-Link has a virtual com port feature, and normally this is connected to USART2 of the STM32s. So you can use it by using the standard "Serial" object and the build-flag:

`build_flags =   -DSERIAL_UART_INSTANCE=2`

You have to remove the USB-Serial related build-flags.

Kurburis, 2024-01-08:
> If it's a Nucleo64 board then I think it does not have USB serial. The USB is connected to the on-board ST-Link.
>
> The ST-Link has a virtual com port feature, and normally this is connected to USART2 [...]

Thanks, that was the solution! For others that might have the same problem, this was the platformio.ini code I've used:

```
[env:nucleo_f303re]
platform = ststm32
board = nucleo_f303re
framework = arduino
lib_archive = false
build_flags = 
            -DSIMPLEFOC_STM32_DEBUG
            -DSERIAL_UART_INSTANCE=2
monitor_speed = 9600
monitor_port = COM8
lib_deps = askuric/Simple FOC@^2.3.2
            Wire
            SPI
```

Kurburis, 2024-01-08:
> The `lib_archive = false` is definately needed. If you do a make clean, it will solve the issue of not finding the correct PWM driver.
>
> However there may be further issues with the 6PWM [...]

I think the problem is related to this issue: https://github.com/simplefoc/Arduino-FOC/issues/220
The last 3 lines of my serial monitor look like this:
```
TIM15-CH2 TIM16-CH1N TIM8-CH2 TIM2-CH3 TIM16-CH1 TIM1-CH2 score: -6
TIM15-CH2 TIM16-CH1N TIM8-CH2 TIM2-CH3 TIM16-CH1 TIM2-CH3 score: -4
STM32-DRV: no workab
```
I've selected pins so none of them uses the N channel and they are all on different timers, but I still get a negative score:

```
TIM15-CH2 TIM8-CH1 TIM3-CH2 TIM2-CH3 TIM16-CH1 TIM1-CH2 score: -7
```

How is the score calculated?

runger1101001, 2024-01-09:
> How is the score calculated?

It's a fairly complicated question, but basically the more different timers are used, the worse (higher) the score. And a negative score always means some kind of error, the configuration can't be made to work.

-7 means: the high and low low side of one or more phases are not on the same timer. 

When using 6-PWM, it is best to use all the pins from TIM1 or TIM8, using the inverted channels for the low side. 

If you can't do this, the next-best option is to use pins where high and low side are always on the same timer.

Kurburis, 2024-01-10:
> It's a fairly complicated question, but basically the more different timers are used, the worse (higher) the score. And a negative score always means some kind of error, the configuration can't be [...]

Thank you, that was the solution. For each pair of high and low pins I used the same timer and channel (but with the N sufiix for the low one) and that worked.

closed 2024-01-10 by Kurburis