simplefoc/Arduino-FOC #366: [BUG] OpenOCD init failed while using specific pin

Kurburis, 2024-01-10:
This bug happened while trying to solve this issue:  https://github.com/simplefoc/Arduino-FOC/issues/364
**Bug description**
BLCDDriver6PWM function needs 6 pins, With a specific pin I am able to build and upload the code to the board once, but after that it is not possible to upload any code via platformio. To return the board to normal, I have to upload any code via STM32 Cube program.

**Hardware setup**
Board: STM32 Nucleo F303RE
Pin that causes the issue: PA13
Pin functions: [https://i.imgur.com/xuleuID.png](url)**

**Code**
Code that causes the bug:
``` BLDCDriver6PWM driver = BLDCDriver6PWM(PA9, PB0, PB14, PB15, PA12, PA13); ```
It does not depend on the argument position.

Error message:
``` 

xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Error: init mode failed (unable to connect to the target)
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
```

**IDE**
- Platformio

runger1101001, 2024-01-10:
What does your platformio.ini file look like, if I may ask?

Kurburis, 2024-01-10:
> What does your platformio.ini file look like, if I may ask?

```
[env:nucleo_f303re]
platform = ststm32
board = nucleo_f303re
framework = arduino
lib_archive = false
build_flags = 
            -DSIMPLEFOC_STM32_DEBUG
            -DSERIAL_UART_INSTANCE=2
monitor_speed = 115200
monitor_port = COM8
lib_deps = askuric/Simple FOC@^2.3.2
            Wire
            SPI
```

Candas1, 2024-01-10:
Of course, you shouldn't use pa13 or pa14

Kurburis, 2024-01-10:
> Of course, you shouldn't use pa13 or pa14

Are they reserved for something I am not aware of?

Candas1, 2024-01-10:
Yes, for programming the board 😂
https://youtu.be/jEz0C2bT2M0?si=82EXkqy8ii4tdhug

Kurburis, 2024-01-10:
> Yes, for programming the chip 😂 https://youtu.be/jEz0C2bT2M0?si=82EXkqy8ii4tdhug

Thanks!

closed 2024-01-10 by Kurburis