simplefoc/Arduino-FOC #376: Can't change PWM frequency - 550hz [BUG]

GK-383, 2024-02-02:
**Describe the bug**
I am trying in every single point of the library and my code to change the pwm frequency but it doesnt change.. The motor screems really hard at 550hz.. 

**Describe the hardware setup**
I am using a Maple mini (stm32F103CBT6) board controlling a drv8313.. 12v battery and big gimbal motor 2.6ohms coils.. 36N40P..
No matter where I set the PWM frequency.. it just doesnt change the output (550hz).

I am using vMicro to compile the project. Below there is my last try using only the driver class to command it.. I tried modifying the hardware_especific class for stm32 but it did not changed anything.. At the osciloscope I can see the pwm duty change.. but the frequency is always in 550hz

If someone could help would be nice!

<code>
// Open loop motor control example
#include <SimpleFOC.h>

//#define _STM32_DEF_

// BLDC motor & driver instance
// BLDCMotor motor = BLDCMotor(pole pair number);
BLDCMotor motor = BLDCMotor(20);
// BLDCDriver3PWM driver = BLDCDriver3PWM(pwmA, pwmB, pwmC, Enable(optional));
BLDCDriver3PWM driver = BLDCDriver3PWM(PA1, PA2, PA3, PA0);

// Stepper motor & driver instance
//StepperMotor motor = StepperMotor(50);
//StepperDriver4PWM driver = StepperDriver4PWM(9, 5, 10, 6,  8);

//target variable
float target_velocity = 0.5;

// instantiate the commander
Commander command = Commander(Serial);
void doTarget(char* cmd) { command.scalar(&target_velocity, cmd); }
void doLimit(char* cmd) { command.scalar(&motor.voltage_limit, cmd); }
void onMotion(char* cmd) { command.motion(&motor, cmd); }

void setup() {

	// driver config
	// power supply voltage [V]
	driver.voltage_power_supply = 12;
	// limit the maximal dc voltage the driver can set
	// as a protection measure for the low-resistance motors
	// this value is fixed on startup
	driver.voltage_limit = 3;
	driver.pwm_frequency = 30000;
	driver.init();
	// link the motor and the driver
	//motor.linkDriver(&driver);

	// limiting motor movements
	// limit the voltage to be set to the motor
	// start very low for high resistance motors
	// current = voltage / resistance, so try to be well under 1Amp
	//motor.voltage_limit = 4;   // [V]

	// open loop control config
	//motor.controller = MotionControlType::velocity_openloop;
	// init motor hardware
	//motor.init();
	//motor.initFOC();
	// add target command T
	command.add('T', doTarget, "target velocity");
	command.add('L', doLimit, "voltage limit");
	command.add('M', onMotion, "motion control");

	Serial.begin(115200);
	Serial.println("Motor ready!");
	Serial.println("Set target velocity [rad/s]");
	_delay(1000);
	driver.enable();
	driver.setPwm(1, 2, 3);
}

void loop() {

	// open loop velocity movement
	// using motor.voltage_limit and motor.velocity_limit
	//motor.loopFOC();
	//motor.move(target_velocity);
	delay(1);

	// user communication
	//command.run();

runger1101001, 2024-02-02:
Hey, I don't know about vMicro compilation, but your error sounds exactly like what happens if you are using PlatformIO and you don't put `lib_archive = false` in your platformio.ini

Are you using PlatformIO?

If not, do you know if the vMicro compiler correctly deals with weakly bound functions?

GK-383, 2024-02-02:
> Hey, I don't know about vMicro compilation, but your error sounds exactly like what happens if you are using PlatformIO and you don't put `lib_archive = false` in your platformio.ini
> 
> Are you using PlatformIO?
> 
> If not, do you know if the vMicro compiler correctly deals with weakly bound functions?

Hello, 

I wasn't able to find anything about plataformIO inside visual micro.. This compiler is a Tool inside Visual Studio, it can compile using its own compiling method or it can use the arduino itself and using both methods does not change the result.. all the .ino files appears at the solution manager.. I can select any arduino project and compile inside it.. 

But I really can't understand why the pwm frequency is not being able to change.. Any other project without simpleFOC where I control the frequencies by myself inside the code it responds exactly how it supposed to.

About weakly bound functions I might not have the necessarry knowledge to awnser properly. https://www.visualmicro.com/

runger1101001, 2024-02-05:
You have to select the correct board in ArduinoIDE, and then it will compile with the correct architecture... If it's compiling for ESP32, it's not gonna work for STM32 MCUs.

In order to target STM32 boards, you have to install the stm32duino board manager package. You have to add the URL for it in the settings.

GK-383, 2024-02-05:
> You have to select the correct board in ArduinoIDE, and then it will compile with the correct architecture... If it's compiling for ESP32, it's not gonna work for STM32 MCUs.
>
> In order to target [...]

I am compiling for Maple Mini Board, it uses STM32.. Any code/library is working for it, except simpleFOC.

If cut out all the hardware directories but the ESP32 it can compile the code.. Meaning it is not using the STM32 files.. 

[image]

runger1101001, 2024-02-05:
You can’t use the maple framework. You have to use stm32duino. The library is not compatible with maple framework.

runger1101001, 2024-02-05:
If it's ok, I'm going to close this bug. There's no plans to support rogerclarkemelbourne's core at the moment I'm afraid.

stm32duino is the official core, and well supported.

There are presumably various ways you can get this board working with stm32duino, as there will be (at least) generic board definitions for this MCU.

But how to get this done is out of scope for our GitHub issues, and I would ask you to get advice from another source such as for example the Arduino Forums, stm32duino forums or maybe the SimpleFOC forums someone can help.

closed 2024-02-05 by runger1101001