simplefoc/Arduino-FOC #377: [BUG] Monitor downsampling cannot be disabled

sylque, 2024-02-06:
**Describe the bug**
There seems to be no way to disable monitor downsampling. Setting `monitor_downsample` to zero disables monitoring altogether.

The reason lies [here](https://github.com/simplefoc/Arduino-FOC/blob/master/src/common/base_classes/FOCMotor.cpp#L94):
```c++
if( !monitor_downsample || monitor_cnt++ < monitor_downsample ) return;
```
This should be replaced by:
```c++
if(monitor_cnt++ < monitor_downsample ) return;
```

**Describe the hardware setup**
ESP32 + SimpleFOCMini + AS4048A + miniatute gimbal motor

**IDE you are using**
Platformio

askuric, 2024-04-19:
This would be a breaking change as down-sampling of 0 is intentionally intended to seamlessly disable monitoring. So many users have used this way in the past. (including me :D) 
So what I propose is to do this :
```cpp
  if( !monitor_downsample || monitor_cnt++ < (monitor_downsample-1) ) return;
```
Now down-sample of 1 actually disables down-sampling.

sylque, 2024-04-19:
Looks great, thanks.

closed 2024-04-23 by askuric