simplefoc/Arduino-FOC #386: driverAlign() ignores gain sign allways "MOT: Success: 3"

4ndreas, 2024-02-29:
I try to get current sensing working on a custom STM32G431CUBx board. With low side current sensing based on three shunt resistors and the internal op amps.

So far the motor spins fine without current sensing.
Also I tested the CS with a DC current on each phase to calibrate the Gain and verify the hardware -> all ok.

But the dirverAlign() funktion always returns  `MOT: Success: 3`

I set the Gain to "70" and stepp thru the calibration 

ab_ratio and ac_ratio are about ~2

[image]

at the end of the function all gains are <0 ->`MOT: Success: 3`

[image]

But if I invert the gain  to "-70"

the currents are inverted and the ratio is ~2 so:

[image]

this changes the gains again:
-> `Success: 3`

[image]

So something is wrong I'm not where the problem is, my gain should be positive since if I disconnect the motor and  input a DC current to on phase the ADC value goes up. To test this I just enabled the low side mosfet and hocked up a second power supply ind constant current mode. 

Also my Currents are off by a factor of 10 my opamp gain is 7 but then the measured current values are by 10 to high (but this is probably a different problem) 
The DC supply is set to 1A and the ADC values go up by like ~80 digits my shunt resistor is 0.01 Ohms. 

**IDE you are using**
- Arduino IDE

Candas1, 2024-02-29:
Hi,

This is how the code works.
It will just give you success 3 if the gains are negative, even if it's correct.

That's not a problem but it could be improved.

4ndreas, 2024-02-29:
Thank you, it is a little bit confusing and I was not sure if the output was intended. 
It would help, if you change the sign of the gain it should somehow influence the outcome downstream ...

askuric, 2024-07-22:
Hi guys, 
The new current sense init #422 is in the release v2.3.4. 
It has several new features and that make it easier to use and has the suggested modification.  :D
The new code returns 3 or 4 only if the gains are actually inverted.

closed 2025-07-26 by askuric