simplefoc/Arduino-FOC #390: Support for the new ESP32 LEDC library API

flowersauce, 2024-03-21:
In the arduino-ESP32 v3.0.0, there have been changes to the LEDC library API. I kindly request your support in updating the simpleFOC library to accommodate these changes.
[image]
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

askuric, 2024-03-21:
I'd like to keep this issue open until we actually do support the v3.0.0!

runger1101001, 2024-03-21:
It seems to me its still an alpha release?

It hasn't made it into the regular esp32 support of PlatformIO yet either. It was a bit of a pain to get it working, but after finding the right URL for the tasmota version of the framework I have PlatformIO compiling and running code on ESP32-C6 :-) 

As far as I am aware Arduino IDE users also still have to reference the dev version of the boards manager URL?

So while I completely agree that we need to work on the support for it, and soon, I think we have to choose the moment to merge it to the main release carefully, so as not to break compatibility for a lot of users.

askuric, 2024-07-22:
In the new release v2.3.4

closed 2024-07-22 by askuric