simplefoc/Arduino-FOC #391: Is this FOC[FEATURE]

jnxyatmjx, 2024-04-02:
**Is your feature request related to a problem? Please describe.**
Description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Description of what you want to happen.

**Describe alternatives you've considered**
Description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

runger1101001, 2024-04-02:
Dear @jnxyatmjx ,

Our users have run motors in excess of 40000RPM, so achieving 5000RPM is possible.

It depends on the correct choice of hardware including sensor, driver and MCU. In particular, you're unlikely to achieve such high speeds on a slow MCU like the Arduino UNO, you'll have to pick something faster.

If its ok, I will close this issue now.

closed 2024-04-02 by runger1101001

jnxyatmjx, 2024-04-16:
> Dear @jnxyatmjx ,
>
> Our users have run motors in excess of 40000RPM, so achieving 5000RPM is possible.
>
> It depends on the correct choice of hardware including sensor, driver and MCU. In particular, [...]

clear , thank you