simplefoc/Arduino-FOC #417: [FEATURE] Possibility to run FOC from interrupt

Candas1, 2024-07-06:
Some MCUs are fast enough to run the FOC algorithms from the current sense interrupt.
This was already discussion [here](https://community.simplefoc.com/t/running-foc-control-loop-inside-to-pwm-timer-overflow-interrupt-handler-or-adc-sample-finish-interrupt-handler/723/7) in the forum, but it was 3 years ago, maybe the opinions have changed.

## This could have several benefits:
### Constant dt:
FOC will run at constant dt, this will simplify PID and Filter math and help with performance as there is no need to measure the timestamp and calculate dt ([this](https://github.com/simplefoc/Arduino-FOC/issues/325) would be needed).

### Better timing:
It makes sure FOC runs on recent phase current data, and duty cycles is applied within the same pwm cycle.

### Overcurrent protection
This can be used for cycle by cycle software overcurrent protection.

### Sensorless
It allows injection for HFI sensorless algorithm.
HFI also needs to identify V1 and V7, and run different code for each with low side current sensing.

## Possible solutions:
### Callback
Allow the user to set up a callback function that will be executed in the current sense interrupt. User should manage the callback to run loopfoc() and move().

### Managed by the library
On MCUs with a FPU (e.g. based on STM32 family) , loopfoc() and move() are directly run from the interrupt.
We should make sure the user is not running loopfoc() and move() from the main loop, or it shouldn't have any effect.

Risks:
- the code should not overrun, a flag could be used to detect this, or this interrupt should be disabled before FOC and enabled after FOC.

nmscode, 2024-07-06:
An example implementation of this can be seen in https://github.com/mcells/Arduino-FOC/tree/esp32-hfi-flux_observer

that branch has several other features but the basic callback is there as well. 

This would require some changes to current sense on stm32 and esp32 to work. From my testing it does work and is quite useful particularly the ocp.

Copper280z, 2024-07-06:
One complication is SPI/I2C encoder updating for sensored applications. On ARM chips I can confirm that SPI reads from an ISR work fine presuming there's sufficient time for it to happen, but I'm unsure that's true for any other architecture (AVR/Xtensa/RISCV). 

Still it might be advisable to not use the ADC ISR if there's an SPI/I2C sensor. I can't think of a good way to enforce that reliably, so I think that means ISR vs loop execution should be user configurable?

nmscode, 2024-07-06:
> Still it might be advisable to not use the ADC ISR if there's an SPI/I2C sensor. I can't think of a good way to enforce that reliably, so I think that means ISR vs loop execution should be user configurable?

Agreed that the isr vs loop execution should be a flag or otherwise user configurable on mcus that support it.