simplefoc/Arduino-FOC #418: [FEATURE] Hall sensor without interrupt

Candas1, 2024-07-06:
I wrote [this](https://github.com/Candas1/Arduino-FOC/commit/d7d44887737f6dd949c2fc06bb7a153d9b026b36) code some time ago but I haven't published it because I wasn't sure it was useful enough. I am posting it here to get your opinion, specially you @dekutree64  as you did a lot of work on this sensor.

What it does is:
- if no interrupts are enabled for hall sensors, it will just check the hall states directly when sensor is updated.
- some small code optimizations

The user can choose if he wants to run it with or without interrupts.

Benefits:
- if user forgets to enable interrupts, the motor will still run (we could issue a warning in debug mode)
- if there are not enough interrupts available (arduino?), the motor can still run
- With interrupts, the faster the motor spins the more interrupts are triggered, slowing down the loop duration/frequency. Without interrupts, the code runs at constant duration/frequency
- PIDs/filters might not be impacted by hall interrupts running in the middle
- less problems when there are noise issues ? (to be checked)

Downside:
- the slower the MCU is and the faster the motor spins, the less accurate the position will be

dekutree64, 2024-07-07:
Looks good to me.

I don't think you can drive faster than one update per sector anyway, since the lagging stator field will start decelerating the rotor instead of accelerating it.

You have to miss at least 2 pulses before the position is lost, so it would require some fast backdriving before it's a problem. The main place I can see that happening is rolling down a hill on an electric skateboard, and that doesn't need to keep track of absolute position.

The main thing you'll lose is accurate timestamps for velocity calculation. May need higher lowpass Tf to average it out, resulting in slower velocity PID response time. Probably not enough to be a problem, though.

Candas1, 2024-07-07:
Thanks a lot for your very valid comments.
I am getting 320 kv motors with hall sensors soon, I can test this at high speeds with slow and fast MCUs and compare the velocity with and without interrupts.

This is more like a fallback mode but I want to make sure it works.

dekutree64, 2024-07-15:
This is awesome. I can use 41F halls now, which don't need any filtering capacitors. They generate excessive interrupts even if I do add capacitors, but work effortlessly with this new code.

I'm getting about 190rad/s top speed. Loop time is around 400us, so 2.5kHz. With 7 pole pairs that's 2500/7/6=59.5 revolutions per second, or 374rad/s theoretical top speed. I don't think nyquist frequency applies here, so I'm not sure why it's maxing out at half, but IIRC that's been about my top speed with other configurations too.

Candas1, 2024-07-15:
Happy to hear it's useful, I will check if I can get it in the next release.

Candas1, 2024-07-21:
This was released. Thanks again for the feedback.

closed 2024-07-21 by Candas1