simplefoc/Arduino-FOC #427: Standalone library usage without Arduino

JeyRunner, 2024-07-22:
Hi, I really like this project but its dependency on the Arduino library currently hinders me from using it.
I know the project name is `Arduino-FOC` but I was wondering if it would be possible to use parts of this library without the Arduino dependency and just cmake as the build system.

The use case is that I want to use this library with another hardware abstraction framework that is more low-level than Arduino, e.g. just stm32 HAL or [modm](https://github.com/modm-io/modm).

Are there already attempts for doing this or does someone have an idea how this could be done?

Candas1, 2024-07-22:
Hi,

@runger1101001 and @askuric might have a better answer, but here are my 2 cents.
SimpleFOC is made to be simple, so it's using the arduino abstraction layer to call STM HAL/ESP IDF/... from Drivers (PWM, current sense, sensors, ....)
For STM32 you could use the STM Motor Workbench, it's also generating the motor control code for you using STM HAL or even low level code.
Modm looks amazing but it's a totally different paradigm.

Someone started a few months ago to port SimpleFOC to Rust, and I believe he had to spend a lot of time learning motor control theory first.
I suggest you try SimpleFOC to learn about Motor Control, before trying to port to something like Modm.

Candas1, 2024-07-22:
Let's take an example.
You want to sample phase current with pin PA0.
The arduino framework allows you to use [pinmaps](https://github.com/stm32duino/Arduino_Core_STM32/blob/73e0c8460e665044e1e91c26e925546a5e226029/variants/STM32F1xx/F103R(4-6)T/PeripheralPins.c#L36) for each variant so that you don't need to know which ADC and which ADC channel must be used.
You cannot do this with STM HAL. Can modm do that ?

Candas1, 2024-07-23:
I think the library now relies on the arduino framework in many different areas:
- pwm drivers use pinmode and hardware timers from arduino
- current sense drivers are mostly using HAL except pinmode and some pinmap usage
- sensors must be using digital read/interrupts/SPI/I2C...
- serial is used for commander/debug communication, maybe it's not a problem with SIMPLEFOC_DISABLE_DEBUG [build flag](https://docs.simplefoc.com/cheetsheet/build_flags)

You could stripping the library from all the drivers you don't need and focus on pwm/open loop for now.

JeyRunner, 2024-07-30:
Thanks @Candas1.
This sounds reasonable (removing all Arduino-dependent parts you mentioned to get a minimal open-loop version).

Candas1, 2024-07-30:
> Thanks @Candas1.
> This sounds reasonable (removing all Arduino-dependent parts you mentioned to get a minimal open-loop version).

I am not saying we will do it, there are other priorities. 
If you want to experiment you can try it.

JeyRunner, 2024-07-30:
Yes I meant that I would try it (I was not expecting that someone else would attempt this, since this is a small use case :sweat_smile: )

Candas1, 2024-07-30:
OK I missunderstood 😂
But feel free to ask if you have any questions

GibbyChan, 2025-04-09:
May I ask the reason why not try ODrive? I'm considering to do similar things recently.