simplefoc/Arduino-FOC #428: [BUG] Cannot compile with esp32-c5

ssaattww, 2024-07-28:
**Describe the bug**
Build error occurs on both Arduino IDE and Platformio
Additionally, it happens with both versions 2.3.3 and 2.3.4.

**Describe the hardware setup**
- Which microcontroller
  - esp32-c3(m5stamp-c3)
- Which position sensor
  -  AS5600
- Current sensing used?
  - not used, I have no plans to use it, how can I disable it?

**IDE you are using**
- Arduino IDE 
- Platformio

**Tried the Getting started guide? - if applicable**
Yes

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32c3]
platform = espressif32
board = esp32-c3-devkitm-1
framework = arduino
monitor_speed = 115200
; board_build.partitions = no_ota.csv
lib_deps = 
	askuric/Simple FOC@^2.3.3
lib_archive = false

```

```cpp
#include <SimpleFOC.h>
// #include <M5Unified.h>

MagneticSensorI2C sensor = MagneticSensorI2C(AS5600_I2C);
TwoWire I2Cone = TwoWire(0);

//Motor parameters
BLDCMotor motor = BLDCMotor(7);
BLDCDriver3PWM driver = BLDCDriver3PWM(4,5,6);

//Command settings
float target_velocity = 0;
Commander command = Commander(Serial);
void doTarget(char* cmd) { command.scalar(&target_velocity, cmd); }

void setup() {
  // use monitoring with serial 
  Serial.begin(115200);
  // enable more verbose output for debugging
  // comment out if not needed
  SimpleFOCDebug::enable(&Serial);

  // initialise magnetic sensor hardware
  I2Cone.begin(8,9);
  sensor.init(&I2Cone);
  // link the motor to the sensor
  motor.linkSensor(&sensor);

  // driver config
  // power supply voltage [V]
  driver.voltage_power_supply = 12;
  driver.init();
  // link the motor and the driver
  motor.linkDriver(&driver);

  // choose FOC modulation (optional)
  motor.foc_modulation = FOCModulationType::SpaceVectorPWM;

  // set motion control loop to be used
  motor.controller = MotionControlType::angle;

  // contoller configuration
  // default parameters in defaults.h

  // velocity PI controller parameters
  motor.PID_velocity.P = 0.2f;
  motor.PID_velocity.I = 20;
  motor.PID_velocity.D = 0;
  // maximal voltage to be set to the motor
  motor.voltage_limit = 6;

  // velocity low pass filtering time constant
  // the lower the less filtered
  motor.LPF_velocity.Tf = 0.01f;

  // angle P controller
  motor.P_angle.P = 20;
  // maximal velocity of the position control
  motor.velocity_limit = 20;
  
  // comment out if not needed
  motor.useMonitoring(Serial);

  // initialize motor
  motor.init();
  // align sensor and start FOC
  motor.initFOC();

  // add target command T
  command.add('T', doTarget, "target angle");

  Serial.println(F("Motor ready."));
  Serial.println(F("Set the target angle using serial terminal:"));
  _delay(1000);
  
}

void loop() {
  // M5.Log.printf(String(sensor.getAngle()).c_str());
  // M5.Log.printf("\n");
  Serial.print(sensor.getAngle()); 
  Serial.println();
  motor.loopFOC();

  motor.move(target_velocity);
  
  command.run();
}
```

```
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_adc_driver.cpp:141:100: note: #pragma message: SimpleFOC: Using analogRead for ADC reading, no fast ADC configuration available!
 #pragma message("SimpleFOC: Using analogRead for ADC reading, no fast ADC configuration available!")
                                                                                                    ^
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp: In function 'void* _configureADCInline(const void*, int, int, int)':
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp:20:3: error: no matching function for call to 'ESP32CurrentSenseParams::ESP32CurrentSenseParams(<brace-enclosed initializer list>)'
   };
   ^
In file included from .pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp:1:
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note: candidate: 'ESP32CurrentSenseParams::ESP32CurrentSenseParams()'
 typedef struct ESP32CurrentSenseParams {
                ^~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note:   candidate expects 0 arguments, 2 provided
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note: candidate: 'constexpr ESP32CurrentSenseParams::ESP32CurrentSenseParams(const ESP32CurrentSenseParams&)'
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note:   candidate expects 1 argument, 2 provided
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note: candidate: 'constexpr ESP32CurrentSenseParams::ESP32CurrentSenseParams(ESP32CurrentSenseParams&&)'
.pio/libdeps/esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note:   candidate expects 1 argument, 2 provided
```

askuric, 2024-07-29:
Hi @ssaattww,

This is interesting. We've got the CI compile check for esp32-c3 in our github repo and it did pass. 
We are compiling for the generic variant `esp32c3` not the `esp32-c3-devkitm-1` though.
 
Could you try compiling for the `esp32c3` to maybe?
Also our CI is arduino ide base, could you try compiling with Arduino IDE?

In any case, we should investigate this. 

The compile error seems pretty strange though, is should not have compiled for any library version from v2.x if this is the case. 

You can also try to do a quick fix and make it compile you can navigate to these lines in your code:
https://github.com/simplefoc/Arduino-FOC/blob/c72f063f3d2d2e26d49a22a7383310cf8f88134b/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp#L17-L20

And replace them by:
```cpp
ESP32CurrentSenseParams* params = new ESP32CurrentSenseParams {
    .pins = { pinA, pinB, pinC },
    .adc_voltage_conv = (_ADC_VOLTAGE)/(_ADC_RESOLUTION),
    .adc_buffer = {0,0,0},
    .buffer_index  = 0,
    .no_adc_channels =0.
  };
```

Jason2866, 2024-08-10:
WOW, genius to use whitespaces for an library name
```
lib_deps = 
	askuric/Simple FOC@^2.3.3
```

askuric, 2024-08-10:
Yeah, it's a consequence of an unfortunate decision made at the first library release back at the day. :)

Windows should be gine with some quotation marks. :D

irl-ops, 2024-10-08:
I am trying to compile for a WEMOS S2 Mini and am getting the same error.  Unfortunately @askuric's above fix does not work, I just get the same error again.  
```
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp: In function 'void* _configureADCInline(const void*, int, int, int)':
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp:23:3: error: no matching function for call to 'ESP32CurrentSenseParams::ESP32CurrentSenseParams(<brace-enclosed initializer list>)'
   };
   ^
In file included from .pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp:1:
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note: candidate: 'ESP32CurrentSenseParams::ESP32CurrentSenseParams()'
 typedef struct ESP32CurrentSenseParams {
                ^~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note:   candidate expects 0 arguments, 5 provided
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note: candidate: 'constexpr ESP32CurrentSenseParams::ESP32CurrentSenseParams(const ESP32CurrentSenseParams&)'
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note:   candidate expects 1 argument, 5 provided
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note: candidate: 'constexpr ESP32CurrentSenseParams::ESP32CurrentSenseParams(ESP32CurrentSenseParams&&)'
.pio/libdeps/lolin_s2_mini/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.h:14:16: note:   candidate expects 1 argument, 5 provided
*** [.pio\build\lolin_s2_mini\lib371\Simple FOC\current_sense\hardware_specific\esp32\esp32_mcu.cpp.o] Error 1
```

Any help is greatly appreciated, thanks for the otherwise excellent software!

askuric, 2024-10-10:
It is an interesting one, I was unable to replicate this issue.  If anyone has ideas why this could happen, I'm all ears. 

So another thing that you could try is:
```cpp
ESP32CurrentSenseParams* params = new ESP32CurrentSenseParams();
params->pins = { pinA, pinB, pinC };
params->adc_voltage_conv = (_ADC_VOLTAGE)/(_ADC_RESOLUTION);
```

BTW. Rolling back to  2.3.3 version of the library should not work really, as v2.3.3 uses the Arduino-esp32 v2.x and the v2.3.4 uses the v3.x of the arduino-esp32. So when you rolled back did you roll back with the arduino-esp32 package as well?
Which version of the arduino-esp32 did you use when you were compiling for v2.3.4? Was is v3.x?

inb4ohnoes, 2024-10-26:
Popping in to say I'm also experiencing this problem (.pio/libdeps/seeed-xiao-esp32c3/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp:20:3: error: no matching function for call to 'ESP32CurrentSenseParams::ESP32CurrentSenseParams(<brace-enclosed initializer list>)' trying to compile for a Seeed Studio XIAO ESP32C3. Fresh project and everything, first dependency. Installed via platform.io.

Santynolo, 2025-09-18:
Hello, i'm using SimpleFOC version 2.3.5, and Espressif 32 version 6.12.0
The compilation only fails on PlatformIO, on Arduino IDE it compiles successfully.
Here is my .ini file
`[env:esp32-c3-devkitm-1]
platform = espressif32
board = esp32-c3-devkitm-1
board_build.mcu = esp32c3
framework = arduino
lib_deps = askuric/Simple FOC@^2.3.5
lib_archive = no
monitor_speed = 115200
build_flags =
    -D ARDUINO_USB_MODE=1
    -D ARDUINO_USB_CDC_ON_BOOT=1`
Nothing weird i think.
This is the error that it gives on build:
`.pio/libdeps/esp32-c3-devkitm-1/Simple FOC/src/current_sense/hardware_specific/esp32/esp32_mcu.cpp:20:3: error: no matching function for call to 'ESP32CurrentSenseParams::ESP32CurrentSenseParams(<brace-enclosed initializer list>)'`
I tried restructuring the way it's initialized but nothing seems to work. (I'm not very good at C++)

Santynolo, 2025-09-18:
Huh, i didn't know that, thanks.

askuric, 2025-12-12:
Hey @peepo,
The serial could be missing due to many different reasons.
The board is not restarting?
You see no output what se ever?
You can still program it right?