simplefoc/Arduino-FOC #429: [BUG]Noise Issue in open loop velocity Sample with ESP32S3 and DRV8311H

arms22, 2024-08-16:
**Describe the bug**

When running the open-loop velocity sample using the ESP32S3 board, a "clicking" or "grinding" noise can be heard from the brushless motor. The noise sounds like a repetitive "krr krr krr".

Build and upload the following code, then send 'T10' from the Serial Monitor.The noise frequency is also proportional to the rotation speed. At 'T1', the noise occurs 3 times per second, while at 'T3', it occurs 9 times per second.There is no noise when using esp32 core version 2.0.7 and SimpleFOC version 2.3.3.

```
#include <SimpleFOC.h>

// 
// BLDC motor & driver instance
BLDCMotor motor = BLDCMotor(7);
// BLDCDriver3PWM driver = BLDCDriver3PWM(6, 0, 1);
BLDCDriver3PWM driver = BLDCDriver3PWM(17, 18, 47);

//target variable
float target_velocity = 0;

// instantiate the commander
Commander command = Commander(Serial);
void doTarget(char* cmd) {
  command.scalar(&target_velocity, cmd);
}
void doLimit(char* cmd) {
  command.scalar(&motor.voltage_limit, cmd);
}

void setup() {

  // use monitoring with serial
  Serial.begin(115200);
  // enable more verbose output for debugging
  // comment out if not needed
  SimpleFOCDebug::enable(&Serial);

  // driver config
  // power supply voltage [V]
  driver.voltage_power_supply = 12;
  // driver.pwm_frequency = 10000;
  // limit the maximal dc voltage the driver can set
  // as a protection measure for the low-resistance motors
  // this value is fixed on startup
  driver.voltage_limit = 6;
  if (!driver.init()) {
    Serial.println("Driver init failed!");
    return;
  }
  // link the motor and the driver
  motor.linkDriver(&driver);

  // limiting motor movements
  // limit the voltage to be set to the motor
  // start very low for high resistance motors
  // current = voltage / resistance, so try to be well under 1Amp
  motor.voltage_limit = 3;  // [V]

  // open loop control config
  motor.controller = MotionControlType::velocity_openloop;

  // init motor hardware
  if (!motor.init()) {
    Serial.println("Motor init failed!");
    return;
  }

  // add target command T
  command.add('T', doTarget, "target velocity");
  command.add('L', doLimit, "voltage limit");

  Serial.println("Motor ready!");
  Serial.println("Set target velocity [rad/s]");
  _delay(1000);
}

void loop() {
  // open loop velocity movement
  // using motor.voltage_limit and motor.velocity_limit
  // to turn the motor "backwards", just set a negative target_velocity
  motor.move(target_velocity);

  // user communication
  command.run();

  // static uint32_t count = 0;
  // static uint32_t last = 0;
  // count++;
  // uint32_t curr = millis();
  // if (curr - last >= 1000) {
  //   Serial.println(count);
  //   count = 0;
  //   last = curr;
  // }
}
```

**Describe the hardware setup**
Motor: BDUAV2206-260KV
Driver: DRV8311H
Microcontroller: ESP32S3 (ESPr® Developer S3 Type-C)
[Product Link](https://ssci.to/8514)
Position Sensor: None
Current Sensing: None

**IDE you are using**
IDE: Arduino IDE
ESP32 Board Version: v3.0.4
Library Version: SimpleFOC 2.3.4
Compile Settings:
USB CDC On Boot: Enabled
USB Mode: Hardware CDC and JTAG

**Tried the Getting started guide? - if applicable**

arms22, 2024-08-16:
I will attach a video. Please note that it might be difficult to hear on some speakers.

https://github.com/user-attachments/assets/22a38457-f3e1-4df6-9a8d-104b569c4560

https://github.com/user-attachments/assets/67bb8630-5b98-4379-9b98-5ac135d41f6c

arms22, 2024-08-28:
Changing the following eliminated the noise: `driver.voltage_limit = 6 -> 8;`

However, changing the following caused noise: `motor.voltage_limit = 4 -> 4;`

I suspect that the PWM waveform may be distorted around the `driver.voltage_limit` voltage.

arms22, 2024-08-28:
I checked the PWM waveform, but did not find any issues.
The attached image shows the PWM waveforms of the A/B outputs. The noise occurred precisely when the light blue waveform disappeared. The cause might be related to the driver's specifications. Apologies for the confusion.
It seems that setting motor.voltage_limit to less than half of driver.voltage_limit can avoid this issue, so I will proceed with that.
[image]

Candas1, 2024-08-28:
Hi,

Maybe it's related to [this](https://docs.simplefoc.com/foc_theory#which-one-should-i-use)

arms22, 2024-09-03:
Thank you for sharing the relevant links. It seems that Ua, b, c are saturated and exceeding the maximum value.

closed 2024-09-05 by arms22