simplefoc/Arduino-FOC #431: [FEATURE]Dshot & Bi-directional Dshot support

lida2003, 2024-09-09:
**Describe the solution you'd like**
Support Dshot & Bi-directional Dshot like AM32, https://github.com/am32-firmware/AM32/blob/main/Src/dshot.c

Candas1, 2024-09-09:
Hi,

This is a motor control library, I am not sure a communication protocol belongs here.
Ideally someone would create a library for dshot, have you checked if there is one already?
There are libraries for other protocols.

runger1101001, 2024-09-09:
I'm with Candas on this - it's unlikely we would add Dshot support to the core library. 

But we would be happy to add it in our drivers library if the implementation can be made in a fairly simple way without external library dependencies.

Otherwise, I think a seperate project would be best, which we would be happy to host/fork in the simplefoc organisation.

Note: I think it would be an interesting addition for sure, I have been looking at this and ExpressLRS and it would be cool to have a simple to configure setup for people to use in combination with controlling SimpleFOC.

Candas1, 2024-09-09:
Looks like it requires timers + DMA.
An example here : https://github.com/mokhwasomssi/stm32_hal_dshot
So this would required separate implementation for arduino,stm,esp, ....

lida2003, 2024-09-09:
> it's unlikely we would add Dshot support to the core library.

> This is a motor control library, I am not sure a communication protocol belongs here.

The repo was added a long time ago. When I posted this issue, I hadn't realized that this is a FOC (Field-Oriented Control) library.

> Otherwise, I think a seperate project would be best, which we would be happy to host/fork in the simplefoc organisation.

Currently, AM32 uses sinusoidal control. And I think it would be better to use FOC control. Is there any project(firmware) for Arduino-FOC library used for FPV ESC?

> Looks like it requires timers + DMA.

Yes.

askuric, 2025-07-29:
Hey guys, since this is not something we're gonna implement a priori. I'm gonna close this issue. We can continue the discussion on the forum maybe.

closed 2025-07-29 by askuric