simplefoc/Arduino-FOC #443: [BUG] STM32: linker problem with __attribute__((weak))

NoX209, 2024-11-20:
**Describe the bug**
I'm trying to get the b-g431b-esc1 board with BLDCDriver6PWM working for some weeks.
I can not produce any pwm output.

**Describe the hardware setup**
- No motor
- No driver
- No position sensor
- No Current sensing
- I have tested the following microcontrollers
1. b-g431b-esc1 only with the pins PA8 and PC13 connected to a scope
2. Nucleo STM32F411RE only with the pins PA8 and PB13 connected to a scope

I tested this on 2 development setup:
A. mac m3 ; Platformio core 6.1.16 ; toolchain-gccarmnoneeabi ld 2.40.0.20230627
B. ubuntu 24.04.1 TLS x86_64 ; Platformio core 6.1.16 ; toolchain-gccarmnoneeabi ld 2.40.0.20230627
I use the SimpleFOC version 2.3.3 but also tested version 2.3.4.

**Tried the Getting started guide? **
I have try'd the guide and have a working setup on a ESP32-S3-zero + SimpleFOCmini + AS5048_I2C setup.

My test code:
platformio.ini (example for Nucleo STM32F411RE)
```
[env:nucleo_f411re]
platform = ststm32
board = nucleo_f411re
framework = arduino
lib_deps =
    askuric/Simple FOC@2.3.3
```

main.cpp
```
# include <SimpleFOC.h>

BLDCDriver6PWM driver = BLDCDriver6PWM(
  PA8,PB13,
  PA9,PB14,
  PA10,PB1
);

void setup() {
  Serial.begin(9600);
  SimpleFOCDebug::enable(&Serial);
  int init_result = driver.init();
  Serial.println(init_result);
  driver.enable();
  driver.setPwm(2,3,5);
}

void loop() {
}
```

the results:
1. b-g431b-esc1: no pwm output
2. Nucleo STM32F411RE: no pwm output

The output via the serial console is
```
0
```

I noticed in the time searching for an error, that not the correct "_configure6PWM" function was called, because the weak version from generic_mcu.cpp was used and not the expected on from stm32_mcu.cpp. (i also used Serial.println() ;) )
(see .pio/build/nucleo_f411re/firmware.map)
```
$ cat firmware.map | grep -B1 "_configure6PWM" | head -n2
.pio/build/nucleo_f411re/lib957/libSimple FOC.a(generic_mcu.cpp.o)
.pio/build/nucleo_f411re/lib957/libSimple FOC.a(BLDCDriver6PWM.cpp.o) (_configure6PWM(long, float, int, int, int, int, int, int))
```

I used the following workaround for the STM32 setups, to test my theory:
comment out the function in the file (SimpleFOC/)src/drivers/hardware_specific/generic_mcu.cpp
```
__attribute__((weak)) void* _configure6PWM(long pwm_frequency, float dead_zone, const int pinA_h, const int pinA_l,  const int pinB_h, const int pinB_l, const int pinC_h, const int pinC_l){
```

With the workaround the linker map looks as expected.
```
$ cat firmware.map | grep -B1 "_configure6PWM" | head -n2
.pio/build/nucleo_f411re/lib957/libSimple FOC.a(stm32_mcu.cpp.o)
.pio/build/nucleo_f411re/lib957/libSimple FOC.a(BLDCDriver6PWM.cpp.o) (_configure6PWM(long, float, int, int, int, int, int, int))

```

The serial output now looks also promissing
```
TIM1-CH1 TIM1-CH1N TIM1-CH2 TIM1-CH2N TIM1-CH3 TIM1-CH3N score: 1
TIM1-CH1 TIM1-CH1N TIM1-CH2 TIM1-CH2N TIM1-CH3 TIM3-CH4 score: -6
STM32-DRV: best: TIM1-CH1 TIM1-CH1N TIM1-CH2 TIM1-CH2N TIM1-CH3 TIM1-CH3N score: 1
STM32-DRV: Syncronising timers! Timer no. 1
STM32-DRV: Restarting timer 1
1
```

and now i have PWM output on my STM32 chips.
I don't need to make this workaround on my ESP32 projects.

Could there be a problem with the used STM32 compiler chain?
Can anyone replicate my problem?
It was working once as shown in this repo https://github.com/owennewo/simplefoc_skateboard_g431b

PS: Thanks for this awesome library :)

NoX209, 2024-11-20:
Oh, nice.
YES, with this option i get an PWM signal without using the workaround.
Thanks!

Is this option recommended for SimpleFOC and did i have overlook it?

For others: https://docs.platformio.org/en/latest/manifests/library-json/fields/build/libarchive.html

Candas1, 2024-11-20:
Yes it's mandatory when using simplefoc with platformio, but in one of the later releases of simplefoc it's supposed to be false by default

Candas1, 2024-11-20:
https://github.com/simplefoc/Arduino-FOC/issues/357