simplefoc/Arduino-FOC #445: [BUG] install lib via PlantformIO  miss library.json file, PWM issue.

touchmii, 2025-01-10:
when i use plantformio install simplefoc, its miss library.json file, the "lib_archive" = no configuration is required for PWM to work, and although #357 has fixed this bug, installing libraries via platformio still doesn't work, which can make things difficult for many newcomers.

lib_deps = askuric/Simple FOC@^2.3.4

runger1101001, 2025-01-10:
Actually, we have a library.json file: https://github.com/simplefoc/Arduino-FOC/blob/master/library.json

But it does not work as expected. It covers the compilation of the library code, but you also need to have this option active for the compilation of your application code.

So unfortunately this solution does not solve the configuration issue by itself.

runger1101001, 2025-02-18:
I'm going to close this issue, as I see no way to solve this other than what we already suggest:

Add the line `lib_archive = false` to your *project* platformio.ini. 

It has to be on the compilation of your project, and not just the library.

closed 2025-02-18 by runger1101001