simplefoc/Arduino-FOC #448: [FEATURE] STM32 Generic ADC input

Moddingear, 2025-01-19:
**Is your feature request related to a problem? Please describe.**
Right now, using any sort of current sense on STM32 hogs the corresponding ADCs. Thus, there is no way to add other sensors such as temperature, battery voltage or a simple potentiometer.

**Describe the solution you'd like**
An API to add other sensors to be included in the DMA

**Additional context**
I'm using the STM32 BG431-ESC1 as a test platform,  with Arduino on platformio. analogRead (or any other ADC-related function) won't work.

Moddingear, 2025-01-31:
For those in the same case as me : There's a workaround builtin.

Follow this https://community.simplefoc.com/t/temperature-monitoring-on-b-g431b-esc1/2955/15, except that you should _readADCVoltageLowSide instead of _readADCVoltageInline (as inline uses analogRead). It works for all of the board's inputs

runger1101001, 2025-02-18:
We're working on a new STM32 current sense driver, and this functionality will be part of it. For the moment, the workaround you suggested will work on some boards, I think.

We can leave this issue open until we finish the new current sense driver, but it may take a while, please don't hold your breath.